mirrexagon/nixpkgs-esp-dev #5: `idf.py` can't get Git revision to report version

mirrexagon, 2021-11-09:
Running `idf.py --version` complains that Git can't get the version.

```
$ idf.py --version
fatal: not a git repository: '/nix/store/zxyj5pafs2v9dbc1ij30zxg00cyi20a9-esp-idf-v4.3.1/.git'
WARNING: Git version unavailable, reading from source
ESP-IDF v4.3.1
```

Probably can be ultimately solved by using the `deepClone` option in `fetchFromGitHub` in `pkgs/esp-idf/default.nix` (just `leaveDotGit` doesn't work because no tags are present I think, Git "cannot describe anything") and making sure to copy the `.git` directory to `$out`, but that causes problems fetching/checking out the repository.

This would change all the esp-idf hashes, so they have to be updated in:
- `pkgs/esp-idf/default.nix`
- `examples/shell-override-versions.nix`

Needs more investigation. We still do get a version number regardless at the moment.

mirrexagon, 2023-06-28:
There is also a similar problem when building a project - cannot use `git describe` to get version information. I'm not sure if that is related to this, since the latter presumably is about the project, not ESP-IDF itself.

knightpp, 2023-09-08:
I cannot build rust project because it complains about absence of git repository. And I think I cannot force it to skip the check.

```
  --- stderr
  Build configuration: BuildConfig {
      esp_idf_tools_install_dir: None,
      esp_idf_sdkconfig: None,
      esp_idf_sdkconfig_defaults: None,
      mcu: Some(
          "esp32c3",
      ),
      native: NativeConfig {
          esp_idf_version: Some(
              Tag(
                  "v5.1",
              ),
          ),
          esp_idf_repository: None,
          esp_idf_cmake_generator: None,
          idf_path: Some(
              "/nix/store/g2s0fiybk9y5p9yiw9v3yvp2gyn7xi98-esp-idf-v5.1",
          ),
          extra_components: [],
          esp_idf_components: None,
      },
      esp_idf_sys_root_crate: None,
  }
  fatal: not a git repository (or any of the parent directories): .git
  fatal: not a git repository (or any of the parent directories): .git
  Error: '/nix/store/g2s0fiybk9y5p9yiw9v3yvp2gyn7xi98-esp-idf-v5.1' is not a git respository

  Caused by:
      0: command 'cd "/nix/store/g2s0fiybk9y5p9yiw9v3yvp2gyn7xi98-esp-idf-v5.1" && LANGUAGE="" LC_ALL="C.UTF-8" "git" "rev-parse" "--show-toplevel"' exited with non-zero status code 128
      1: fatal: not a git repository (or any of the parent directories): .git
```

simisimis, 2023-10-21:
I entered the shell through `nix develop github:mirrexagon/nixpkgs-esp-dev#esp32c6-idf`
then did `cargo generate esp-rs/esp-idf-template cargo`
cd to a generated project and tried doing `cargo build` and I am getting to the same `not a git ...` error
It's a pity because this repo is so neatly laid out

torbeneims, 2023-12-25:
I cannot build the hello_world example. Has anyone managed to do that yet? The idf complains about `git describe`: Could not use 'git describe' to determine PROJECT_VER, as mentioned above.

Here's a full log:
```
torben@torbens-xps ~/i/idf-nix (main)> nix develop github:mirrexagon/nixpkgs-esp-dev#esp32-idf

[torben@torbens-xps:~/iot/idf-nix]$ cp $IDF_PATH/examples/get-started/hello_world -r .

[torben@torbens-xps:~/iot/idf-nix]$ cd hello_world/

[torben@torbens-xps:~/iot/idf-nix/hello_world]$ chmod +w .

[torben@torbens-xps:~/iot/idf-nix/hello_world]$ idf.py build
Executing action: all (aliases: build)
Running cmake in directory /home/torben/iot/idf-nix/hello_world/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=/nix/store/k13a1nbhdnm607pk33mq3xzirmp11sk9-python3-3.10.12-env/bin/python3.10 -DESP_PLATFORM=1 -DCCACHE_ENABLE=0 /home/torben/iot/idf-nix/hello_world"...
-- IDF_TARGET not set, using default target: esp32
-- Found Git: /nix/store/y0gvg44jdsbn8hnnr27ixjf102nk7a9x-git-2.41.0/bin/git (found version "2.41.0") 
-- git rev-parse returned 'fatal: not a git repository (or any of the parent directories): .git'
fatal: not a git repository (or any of the parent directories): .git
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /nix/store/804azw62j381cvy6c84dhrrxhlby1ly6-xtensa-esp32-elf-esp-idf-v5.1.2/bin/xtensa-esp32-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /nix/store/804azw62j381cvy6c84dhrrxhlby1ly6-xtensa-esp32-elf-esp-idf-v5.1.2/bin/xtensa-esp32-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /nix/store/804azw62j381cvy6c84dhrrxhlby1ly6-xtensa-esp32-elf-esp-idf-v5.1.2/bin/xtensa-esp32-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at build/CMakeFiles/git-data/grabRef.cmake:48 (file):
  file failed to open for reading (No such file or directory):

    /home/torben/iot/idf-nix/hello_world/build/CMakeFiles/git-data/head-ref
Call Stack (most recent call first):
  /nix/store/zw16dh8ns3brx9xx4jmqb731cr72dxa3-esp-idf-v5.1.2/tools/cmake/third_party/GetGitRevisionDescription.cmake:83 (include)
  /nix/store/zw16dh8ns3brx9xx4jmqb731cr72dxa3-esp-idf-v5.1.2/tools/cmake/third_party/GetGitRevisionDescription.cmake:93 (get_git_head_revision)
  /nix/store/zw16dh8ns3brx9xx4jmqb731cr72dxa3-esp-idf-v5.1.2/tools/cmake/project.cmake:78 (git_describe)
  /nix/store/zw16dh8ns3brx9xx4jmqb731cr72dxa3-esp-idf-v5.1.2/tools/cmake/project.cmake:543 (__project_get_revision)
  CMakeLists.txt:6 (project)

-- Could not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file /home/torben/iot/idf-nix/hello_world/sdkconfig
Compiler supported targets: xtensa-esp32-elf

-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of time_t
-- Check size of time_t - done
-- Found Python3: /nix/store/k13a1nbhdnm607pk33mq3xzirmp11sk9-python3-3.10.12-env/bin/python3.10 (found version "3.10.12") found components: Interpreter 
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS - Success
-- App "hello_world" version: 1
-- Adding linker script /home/torben/iot/idf-nix/hello_world/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script /nix/store/zw16dh8ns3brx9xx4jmqb731cr72dxa3-esp-idf-v5.1.2/components/esp_system/ld/esp32/sections.ld.in
-- Adding linker script /nix/store/zw16dh8ns3brx9xx4jmqb731cr72dxa3-esp-idf-v5.1.2/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /nix/store/zw16dh8ns3brx9xx4jmqb731cr72dxa3-esp-idf-v5.1.2/components/esp_rom/esp32/ld/esp32.rom.api.ld
-- Adding linker script /nix/store/zw16dh8ns3brx9xx4jmqb731cr72dxa3-esp-idf-v5.1.2/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /nix/store/zw16dh8ns3brx9xx4jmqb731cr72dxa3-esp-idf-v5.1.2/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script /nix/store/zw16dh8ns3brx9xx4jmqb731cr72dxa3-esp-idf-v5.1.2/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script /nix/store/zw16dh8ns3brx9xx4jmqb731cr72dxa3-esp-idf-v5.1.2/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script /nix/store/zw16dh8ns3brx9xx4jmqb731cr72dxa3-esp-idf-v5.1.2/components/soc/esp32/ld/esp32.peripherals.ld
fatal: not a git repository (or any of the parent directories): .git
fatal: not a git repository (or any of the parent directories): .git
-- Components: app_trace app_update bootloader bootloader_support bt cmock console cxx driver efuse esp-tls esp_adc esp_app_format esp_coex esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_server esp_https_ota esp_https_server esp_hw_support esp_lcd esp_local_ctrl esp_mm esp_netif esp_netif_stack esp_partition esp_phy esp_pm esp_psram esp_ringbuf esp_rom esp_system esp_timer esp_wifi espcoredump esptool_py fatfs freertos hal heap http_parser idf_test ieee802154 json log lwip main mbedtls mqtt newlib nvs_flash openthread partition_table perfmon protobuf-c protocomm pthread sdmmc soc spi_flash [... cut ...] w16dh8ns3brx9xx4jmqb731cr72dxa3-esp-idf-v5.1.2/components/wifi_provisioning /nix/store/zw16dh8ns3brx9xx4jmqb731cr72dxa3-esp-idf-v5.1.2/components/wpa_supplicant /nix/store/zw16dh8ns3brx9xx4jmqb731cr72dxa3-esp-idf-v5.1.2/components/xtensa
-- Configuring incomplete, errors occurred!
cmake failed with exit code 1, output of the command is in the /home/torben/iot/idf-nix/hello_world/build/log/idf_py_stderr_output_97101 and /home/torben/iot/idf-nix/hello_world/build/log/idf_py_stdout_output_97101
```

cyber-murmel, 2024-02-15:
I don't know if this is applicable to other build error, but I notices that `idf.py` fails if it's run in a git repo (i.e. `.git` in a parent dir) that doesn't have any commits yet.
My workaround was to delete the `.git` dir to add it later, when I'm ready to commit.

cyber-murmel, 2024-02-19:
I tried to reproduce this on manjaro, following the official install instructions, but there it seems to work fine in an empty git repo.

jonahbron, 2024-03-10:
I filed a bug against this upstream in esp-idf, it might be something that could make sense to fix there, even if we can get a workaround on this side.

https://github.com/espressif/esp-idf/issues/13345

mirrexagon, 2024-03-17:
@torbeneims 

> I cannot build the hello_world example. Has anyone managed to do that yet? The idf complains about `git describe`: Could not use 'git describe' to determine PROJECT_VER, as mentioned above.
> 
> Here's a full log:
> 
> [...]

Mostly the Git errors don't actually seem to stop the build, but in your case CMake throws an error that might be what stops the build:

```
CMake Error at build/CMakeFiles/git-data/grabRef.cmake:48 (file):
  file failed to open for reading (No such file or directory):

    /home/torben/iot/idf-nix/hello_world/build/CMakeFiles/git-data/head-ref
Call Stack (most recent call first):
  /nix/store/zw16dh8ns3brx9xx4jmqb731cr72dxa3-esp-idf-v5.1.2/tools/cmake/third_party/GetGitRevisionDescription.cmake:83 (include)
  /nix/store/zw16dh8ns3brx9xx4jmqb731cr72dxa3-esp-idf-v5.1.2/tools/cmake/third_party/GetGitRevisionDescription.cmake:93 (get_git_head_revision)
  /nix/store/zw16dh8ns3brx9xx4jmqb731cr72dxa3-esp-idf-v5.1.2/tools/cmake/project.cmake:78 (git_describe)
  /nix/store/zw16dh8ns3brx9xx4jmqb731cr72dxa3-esp-idf-v5.1.2/tools/cmake/project.cmake:543 (__project_get_revision)
  CMakeLists.txt:6 (project)
```

If I run the same commands you did on NixOS, I get a successful build, and that file `git-data/head-ref` it complains about in your log doesn't exist for me at all after the build. Unfortunately I don't really know how this works - I'm looking through the other comments and related issues and PRs now, which might have some clues.

<details>
<summary>My console output from successful build</summary>

```
offline/tmp/idf-nix
❯ nix develop github:mirrexagon/nixpkgs-esp-dev#esp32-idf

[mir@garnet:~/code/offline/tmp/idf-nix]$ cp $IDF_PATH/examples/get-started/hello_world -r .

[mir@garnet:~/code/offline/tmp/idf-nix]$ cd hello_world/

[mir@garnet:~/code/offline/tmp/idf-nix/hello_world]$ chmod +w .

[mir@garnet:~/code/offline/tmp/idf-nix/hello_world]$ idf.py build
Executing action: all (aliases: build)
Running cmake in directory /home/mir/code/offline/tmp/idf-nix/hello_world/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=/nix/store/y6f09a574y2fnl87p446hdiba06xyrqw-python3-3.11.7-env/bin/python3.11 -DESP_PLATFORM=1 -DCCACHE_ENABLE=0 /home/mir/code/offline/tmp/idf-nix/hello_world"...
-- IDF_TARGET not set, using default target: esp32
-- Found Git: /nix/store/za04kin80ybavb7qp577dghpax4mf82z-git-2.43.0/bin/git (found version "2.43.0")
-- git rev-parse returned 'fatal: not a git repository (or any of the parent directories): .git'
fatal: not a git repository (or any of the parent directories): .git
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /nix/store/5wgfn6g2hx0w65yp5win09pgvnh83gd6-xtensa-esp32-elf-esp-idf-v5.1.2/bin/xtensa-esp32-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /nix/store/5wgfn6g2hx0w65yp5win09pgvnh83gd6-xtensa-esp32-elf-esp-idf-v5.1.2/bin/xtensa-esp32-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /nix/store/5wgfn6g2hx0w65yp5win09pgvnh83gd6-xtensa-esp32-elf-esp-idf-v5.1.2/bin/xtensa-esp32-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- git rev-parse returned 'fatal: not a git repository (or any of the parent directories): .git'
-- Could not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file /home/mir/code/offline/tmp/idf-nix/hello_world/sdkconfig
Compiler supported targets: xtensa-esp32-elf

-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
[... 76 lines cut ...]
-- Generating done (0.0s)
-- Build files have been written to: /home/mir/code/offline/tmp/idf-nix/hello_world/build/bootloader
[103/104] Generating binary image from built executableesptool.py v4.6.2
Creating esp32 image...
Merged 1 ELF section
Successfully created esp32 image.
Generated /home/mir/code/offline/tmp/idf-nix/hello_world/build/bootloader/bootloader.bin
[104/104] cd /home/mir/code/offline/tmp/idf-nix/hello_world/build/bootloader/esp-idf/esptool_py && /nix/s...t 0x8000 bootloader 0x1000 /home/mir/code/offline/tmp/idf-nix/hello_world/build/bootloader/bootloader.binBootloader binary size 0x6820 bytes. 0x7e0 bytes (7%) free.
[899/900] Generating binary image from built executableesptool.py v4.6.2
Creating esp32 image...
Merged 2 ELF sections
Successfully created esp32 image.
Generated /home/mir/code/offline/tmp/idf-nix/hello_world/build/hello_world.bin
[900/900] cd /home/mir/code/offline/tmp/idf-nix/hello_world/build/esp-idf/esptool_py && /nix/store/y6f09a.../partition_table/partition-table.bin /home/mir/code/offline/tmp/idf-nix/hello_world/build/hello_world.binhello_world.bin binary size 0x2adf0 bytes. Smallest app partition is 0x100000 bytes. 0xd5210 bytes (83%) free.

Project build complete. To flash, run this command:
/nix/store/y6f09a574y2fnl87p446hdiba06xyrqw-python3-3.11.7-env/bin/python3.11 ../../../../../../../nix/store/qlq76kqz03mah3wamkswpjicqwvyndq3-esp-idf-v5.1.2/components/esptool_py/esptool/esptool.py -p (PORT) -b 460800 --before default_reset --after hard_reset --chip esp32  write_flash --flash_mode dio --flash_size 2MB --flash_freq 40m 0x1000 build/bootloader/bootloader.bin 0x8000 build/partition_table/partition-table.bin 0x10000 build/hello_world.bin
or run 'idf.py -p (PORT) flash'
```

</details>

---

Thank you all for your work and input on this!

anpin, 2024-06-16:
I have somewhat similar issue 

```console
> idf.py set-target esp32
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Build directory '/home/a/projects/esp-tflite-micro/examples/micro_speech/build' not found. Nothing to clean.
Executing action: set-target
Set Target to: esp32, new sdkconfig will be created.
Running cmake in directory /home/a/projects/esp-tflite-micro/examples/micro_speech/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=/nix/store/3f5qi44q5319jbra5fyimj4vk1agiv53-python3-3.11.9-env/bin/python3.11 -DESP_PLATFORM=1 -DIDF_TARGET=esp32 -DCCACHE_ENABLE=0 /home/a/projects/esp-tflite-micro/examples/micro_speech"...
-- Found Git: /nix/store/jxjzgz1p0hdqsg8k17ivw5mgcd6ffx9c-git-2.44.1/bin/git (found version "2.44.1")
-- git rev-parse returned 'fatal: detected dubious ownership in repository at '/nix/store/hh3v33dhnn8xnsp4lgl7y1zybfybx45n-esp-idf-v5.1.3'
To add an exception for this directory, call:

        git config --global --add safe.directory /nix/store/hh3v33dhnn8xnsp4lgl7y1zybfybx45n-esp-idf-v5.1.3'
fatal: detected dubious ownership in repository at '/nix/store/hh3v33dhnn8xnsp4lgl7y1zybfybx45n-esp-idf-v5.1.3'
To add an exception for this directory, call:

        git config --global --add safe.directory /nix/store/hh3v33dhnn8xnsp4lgl7y1zybfybx45n-esp-idf-v5.1.3
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /nix/store/mhfhfsl0fk08lmdc15xf7b12izv1b8y2-xtensa-esp32-elf-esp-idf-v5.1.3/bin/xtensa-esp32-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /nix/store/mhfhfsl0fk08lmdc15xf7b12izv1b8y2-xtensa-esp32-elf-esp-idf-v5.1.3/bin/xtensa-esp32-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /nix/store/mhfhfsl0fk08lmdc15xf7b12izv1b8y2-xtensa-esp32-elf-esp-idf-v5.1.3/bin/xtensa-esp32-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
CMake Error at /nix/store/hh3v33dhnn8xnsp4lgl7y1zybfybx45n-esp-idf-v5.1.3/tools/cmake/build.cmake:540 (message):
  Traceback (most recent call last):

    File "<frozen runpy>", line 198, in _run_module_as_main
    File "<frozen runpy>", line 88, in _run_code
    File "/nix/store/hh3v33dhnn8xnsp4lgl7y1zybfybx45n-esp-idf-v5.1.3/lib/python3.11/site-packages/idf_component_manager/prepare_components/__main__.py", line 6, in <module>
      main()
    File "/nix/store/hh3v33dhnn8xnsp4lgl7y1zybfybx45n-esp-idf-v5.1.3/lib/python3.11/site-packages/idf_component_manager/prepare_components/prepare.py", line 124, in main
      args.func(args)
    File "/nix/store/hh3v33dhnn8xnsp4lgl7y1zybfybx45n-esp-idf-v5.1.3/lib/python3.11/site-packages/idf_component_manager/prepare_components/prepare.py", line 31, in prepare_dep_dirs
      ).prepare_dep_dirs(
        ^^^^^^^^^^^^^^^^^
    File "/nix/store/hh3v33dhnn8xnsp4lgl7y1zybfybx45n-esp-idf-v5.1.3/lib/python3.11/site-packages/idf_component_manager/core.py", line 67, in wrapper
      return func(self, *args, **kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^
    File "/nix/store/hh3v33dhnn8xnsp4lgl7y1zybfybx45n-esp-idf-v5.1.3/lib/python3.11/site-packages/idf_component_manager/core.py", line 529, in prepare_dep_dirs
      downloaded_component_paths, downloaded_component_version_dict = download_project_dependencies(
                                                                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    File "/nix/store/hh3v33dhnn8xnsp4lgl7y1zybfybx45n-esp-idf-v5.1.3/lib/python3.11/site-packages/idf_component_manager/dependencies.py", line 146, in download_project_dependencies
      if is_solve_required(project_requirements, solution):
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    File "/nix/store/hh3v33dhnn8xnsp4lgl7y1zybfybx45n-esp-idf-v5.1.3/lib/python3.11/site-packages/idf_component_manager/dependencies.py", line 99, in is_solve_required
      component.name, spec='=={}'.format(component.version.semver))
                                         ^^^^^^^^^^^^^^^^^^^^^^^^
    File "/nix/store/hh3v33dhnn8xnsp4lgl7y1zybfybx45n-esp-idf-v5.1.3/lib/python3.11/site-packages/idf_component_tools/manifest/manifest.py", line 275, in semver
      raise TypeError('Version is not semantic')

  TypeError: Version is not semantic

Call Stack (most recent call first):
  /nix/store/hh3v33dhnn8xnsp4lgl7y1zybfybx45n-esp-idf-v5.1.3/tools/cmake/project.cmake:547 (idf_build_process)
  CMakeLists.txt:4 (project)

-- Configuring incomplete, errors occurred!
cmake failed with exit code 1, output of the command is in the /home/a/projects/esp-tflite-micro/examples/micro_speech/build/log/idf_py_stderr_output_88446 and /home/a/projects/esp-tflite-micro/examples/micro_speech/build/log/idf_py_stdout_output_88446
[... 42 lines cut ...]
        ^^^^^^^^^^^^^^^^^
    File "/nix/store/hh3v33dhnn8xnsp4lgl7y1zybfybx45n-esp-idf-v5.1.3/lib/python3.11/site-packages/idf_component_manager/core.py", line 67, in wrapper
      return func(self, *args, **kwargs)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^
[... cut ...]
  TypeError: Version is not semantic

Call Stack (most recent call first):
  /nix/store/hh3v33dhnn8xnsp4lgl7y1zybfybx45n-esp-idf-v5.1.3/tools/cmake/project.cmake:547 (idf_build_process)
  CMakeLists.txt:4 (project)

-- Configuring incomplete, errors occurred!
cmake failed with exit code 1, output of the command is in the /home/a/projects/esp-tflite-micro/examples/micro_speech/build/log/idf_py_stderr_output_89444 and /home/a/projects/esp-tflite-micro/examples/micro_speech/build/log/idf_py_stdout_output_89444
```

laxsjo, 2024-09-10:
About my PR: On further inspection it seems like the esp-idf cmake scripts also check for a version.txt in other locations, I haven't had time to check the details, but if we're lucky we it might make it so that the build process never tries to run `git rev-parse`. I should definitely look more into this.

mirrexagon, 2024-10-24:
Whoops, I didn't mean to close this.