mirrexagon/nixpkgs-esp-dev #7: Add support for Rust

wucke13, 2022-02-20:
Hi, I'd like to build code for ESP32 in Rust, on my NixOS machine. I though this would be a good start. Unfortunately, this does not work yet. When I tried the following:

```console
https://github.com/espressif/rust-esp32-example.git
cd rust-esp32-example
idf.py set-target esp32
idf.py build
```
which errors with 

```console
  process didn't exit successfully: `rustc - --crate-name ___ --print=file-names --target xtensa-esp32-espidf --crate-type bin --crate-type rlib --crate-type dylib --crate-type cdylib --crate-type staticlib --crate-type proc-macro --print=sysroot --print=cfg` (exit status: 1)
  --- stderr
  error: Error loading target specification: Could not find specification for target "xtensa-esp32-espidf". Run `rustc --print target-list` for a list of built-in targets
```

This is expected - I did not install the Rust fork yet. However, and this is where I think we might have a problem in the nixpkgs-esp-dev project:

```console
https://github.com/espressif/rust-esp32-example.git
cd rust-esp32-example
idf.py set-target esp32c3
idf.py build
```

results in 

```console
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
-- The ASM compiler identification is unknown
-- Found assembler: riscv32-esp-elf-gcc
CMake Error at /nix/store/zxyj5pafs2v9dbc1ij30zxg00cyi20a9-esp-idf-v4.3.1/tools/cmake/project.cmake:311 (__project):
  The CMAKE_C_COMPILER:

    riscv32-esp-elf-gcc

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "CC" or the CMake cache entry CMAKE_C_COMPILER to the full path to
  the compiler, or to the compiler name if it is in the PATH.
Call Stack (most recent call first):
  CMakeLists.txt:6 (project)

CMake Error at /nix/store/zxyj5pafs2v9dbc1ij30zxg00cyi20a9-esp-idf-v4.3.1/tools/cmake/project.cmake:311 (__project):
  The CMAKE_CXX_COMPILER:

    riscv32-esp-elf-g++

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.
Call Stack (most recent call first):
  CMakeLists.txt:6 (project)

CMake Error at /nix/store/zxyj5pafs2v9dbc1ij30zxg00cyi20a9-esp-idf-v4.3.1/tools/cmake/project.cmake:311 (__project):
  The CMAKE_ASM_COMPILER:

    riscv32-esp-elf-gcc

  is not a full path and was not found in the PATH.

  Tell CMake where to find the compiler by setting either the environment
  variable "ASM" or the CMake cache entry CMAKE_ASM_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.
Call Stack (most recent call first):
  CMakeLists.txt:6 (project)

-- Warning: Did not find file Compiler/-ASM
-- Configuring incomplete, errors occurred!
```

Is this nix version of the toolchain supposed to already work with the esp32c3 (which needs a RISC-V toolchain)?

mirrexagon, 2022-02-20:
No, I haven't added the ESP-specific RISC-V toolchain to this project.

To add it, it's hopefully as easy as copying the ESP32 toolchain derivation and changing the source to the RISC-V toolchain package instead. Under the toolchain releases, the RISC-V toolchain is there along with the various Xtensa ones: https://github.com/espressif/crosstool-NG/releases

mirrexagon, 2022-03-12:
The ESP RISC-V toolchain has been added in #8 - feel free to reopen this issue if you run into more issues.

PR #8 (2022-03-12): feat: add support for RISCV32 toolchains
Hi there, here's a PR that fixes #7 : I do not know what is the optimal way to handle this, rather, maybe in the esp32 toolchain rather than a new toolchain at all.
closed 2022-03-12 by mirrexagon