mirrexagon/nixpkgs-esp-dev #10: Esp32-C3 nix-shell error with python static typing

legendofa, 2022-06-16:
Hi, i am getting the following error when trying to run: `nix-shell esp32c3-idf.nix`.

` ` `sh
building '/nix/store/lcqycb7r7d8288p1gdz3qznm5a80p1pc-py-eval-result.drv'...
Traceback (most recent call last):
  File "<string>", line 2, in <module>
  File "/nix/store/da07biz6d62383s4ykbssysdalxqq3xj-python3.9-packaging-21.3/lib/python3.9/site-packages/packaging/version.py", line 42
    def parse(version: str) -> Union["LegacyVersion", "Version"]:
                     ^
SyntaxError: invalid syntax
error: builder for '/nix/store/lcqycb7r7d8288p1gdz3qznm5a80p1pc-py-eval-result.drv' failed with exit code 1;
       last 6 log lines:
       > Traceback (most recent call last):
       >   File "<string>", line 2, in <module>
       >   File "/nix/store/da07biz6d62383s4ykbssysdalxqq3xj-python3.9-packaging-21.3/lib/python3.9/site-packages/packaging/version.py", line 42
       >     def parse(version: str) -> Union["LegacyVersion", "Version"]:
       >                      ^
       > SyntaxError: invalid syntax
       For full logs, run 'nix log /nix/store/lcqycb7r7d8288p1gdz3qznm5a80p1pc-py-eval-result.drv'.
(use '--show-trace' to show detailed location information)
` ` ` 

My system info:
```console
[user@system:~]$ nix-shell -p nix-info --run "nix-info -m"
 - system: `"x86_64-linux"`
 - host os: `Linux 5.18.3, NixOS, 22.05 (Quokka), 22.05.926.8b538fcb329`
 - multi-user?: `yes`
 - sandbox: `yes`
 - version: `nix-env (Nix) 2.8.1`
 - channels(root): `"nixos-22.05"`
 - nixpkgs: `/nix/var/nix/profiles/per-user/root/channels/nixos`
```

I am on commit 13ba39c98aa1a87384e35eef2e646f9b8521229a.

Full trace:
```sh
building '/nix/store/lcqycb7r7d8288p1gdz3qznm5a80p1pc-py-eval-result.drv'...
Traceback (most recent call last):
  File "<string>", line 2, in <module>
  File "/nix/store/da07biz6d62383s4ykbssysdalxqq3xj-python3.9-packaging-21.3/lib/python3.9/site-packages/packaging/version.py", line 42
    def parse(version: str) -> Union["LegacyVersion", "Version"]:
                     ^
SyntaxError: invalid syntax
error: builder for '/nix/store/lcqycb7r7d8288p1gdz3qznm5a80p1pc-py-eval-result.drv' failed with exit code 1;
       last 6 log lines:
       > Traceback (most recent call last):
       >   File "<string>", line 2, in <module>
       >   File "/nix/store/da07biz6d62383s4ykbssysdalxqq3xj-python3.9-packaging-21.3/lib/python3.9/site-packages/packaging/version.py", line 42
       >     def parse(version: str) -> Union["LegacyVersion", "Version"]:
       >                      ^
       > SyntaxError: invalid syntax
       For full logs, run 'nix log /nix/store/lcqycb7r7d8288p1gdz3qznm5a80p1pc-py-eval-result.drv'.

       … while realising the context of path '/nix/store/rdv4dv1xj7b6zhp87vab1kaszjkljjl5-py-eval-result'

       at /nix/store/zkffps35qh635lbl47zr41b8q4jfmp8c-source/mach_nix/fixes.nix:9:29:

            8|   # evaluate arbitrary python expression and return json parsed result
            9|   py_eval = str: fromJSON ( readFile (
             |                             ^
           10|     pkgs.runCommand "py-eval-result" { buildInputs = with pkgs; [ python3 python3Packages.packaging ]; } ''

       … while evaluating 'py_eval'

       at /nix/store/zkffps35qh635lbl47zr41b8q4jfmp8c-source/mach_nix/fixes.nix:9:13:

            8|   # evaluate arbitrary python expression and return json parsed result
            9|   py_eval = str: fromJSON ( readFile (
[... 349 lines cut ...]
       at /nix/store/fbkb0r9pnafpyq9r43wfz8727h65pi3i-nixos-22.05/nixos/pkgs/stdenv/generic/make-derivation.nix:278:7:

          277|     // (lib.optionalAttrs (attrs ? name || (attrs ? pname && attrs ? version)) {
          278|       name =
             |       ^
          279|         let

       … while evaluating the attribute 'installPhase' of the derivation 'esp-idf-v4.3.1'

       at /nix/store/fbkb0r9pnafpyq9r43wfz8727h65pi3i-nixos-22.05/nixos/pkgs/stdenv/generic/make-derivation.nix:278:7:

          277|     // (lib.optionalAttrs (attrs ? name || (attrs ? pname && attrs ? version)) {
          278|       name =
             |       ^
          279|         let

       … while evaluating the attribute 'buildInputs' of the derivation 'esp-idf'

       at /nix/store/fbkb0r9pnafpyq9r43wfz8727h65pi3i-nixos-22.05/nixos/pkgs/stdenv/generic/make-derivation.nix:278:7:

          277|     // (lib.optionalAttrs (attrs ? name || (attrs ? pname && attrs ? version)) {
          278|       name =
             |       ^
          279|         let
```

mirrexagon, 2022-07-24:
I can reproduce this with:

```
 - system: `"x86_64-linux"`
 - host os: `Linux 5.18.0, NixOS, 22.11 (Raccoon), 22.11.20220530.f1c1676`
 - multi-user?: `yes`
 - sandbox: `yes`
 - version: `nix-env (Nix) 2.8.1`
 - channels(root): `""`
 - nixpkgs: `/nix/store/d84iknazpzwbjbj1j9zh9wnh6v6dzlfb-source`
 ```

The flake-pinned nixpkgs (`nix develop .#esp32c3-idf`) works, but if I update the flake dependencies it breaks in the same way, so I'm guessing something updated in nixpkgs that breaks the mach-nix setup.

I'll see if I can fix it, but until then you can still get a working shell if you have Nix flake support with `nix develop .#esp32c3-idf`

mirrexagon, 2023-06-28:
I think this is the same cause as/would be fixed by a fix for #12, so I am closing this issue and redirecting there.

mirrexagon, 2023-06-30:
#20 should have fixed this.