mirrexagon/nixpkgs-esp-dev #13: Use Python27 to  support vscode debugger

StarGate01, 2023-01-12:
The IDF vscode debugger still depends on Python 2.7: https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/DEBUGGING.md#use-the-esp-idf-debug-adapter

Unfortunately, building for 2.7 is a pain on Nix in general an using mach-nix specifically.

mirrexagon, 2023-06-29:
Do you actually need Python 2.7 in this project (`nixpkgs-esp-dev`) specifically, or can/do you run the debug adapter separately from VSCode? The page you linked maybe implies that VSCode is responsible for launching the debug adapter, and so perhaps the environment created by `nixpkgs-esp-dev` wouldn't directly affect that, unless you launched VSCode inside the environment?

mirrexagon, 2024-10-23:
It seems like that debug adapter is being deprecated and replaced with an Eclipse one (based on what's at the link you posted right now). Also see if #65 might be helpful.

closed 2024-10-23 by mirrexagon