mirrexagon/nixpkgs-esp-dev #16: Flake in project for esp.

Mastergamer433, 2023-01-26:
I have made a flake for the hello world example in espressif's esp-idf repo.
It looks like this:
```nix
{
  description = "Simple get started hello world for esp32";

  inputs = {
    flake-utils.url = "github:numtide/flake-utils";
    nixpkgs.url = "github:nixos/nixpkgs/nixpkgs-unstable";
    nixpkgs-esp-dev.url = "github:mirrexagon/nixpkgs-esp-dev";
  };

  outputs = { self, flake-utils, nixpkgs, nixpkgs-esp-dev }:
    flake-utils.lib.eachDefaultSystem (system:
      let
        pkgs = import nixpkgs-esp-dev;
      in {
        devShells.default = pkgs.mkShell {
          buildInputs = with pkgs; [
            gcc-xtensa-esp32-elf-bin
            esp-idf
            esptool

            # Tools required to use ESP-IDF.
            git
            wget
            gnumake

            flex
            bison
            gperf
            pkgconfig

            cmake
            ninja

            ncurses5
          ];
        };
      });
}
```
but I get his error:
```
error: builder for '/nix/store/5kls4a3vm3as7ybx03vs6d36v719hyll-mach_nix_file.drv' failed with exit code 1;
       last 9 log lines:
       >
       > The Package 'pathlib' (build: None) is not available from any of the selected providers ['nixpkgs', 'sdist', 'wheel']
       >  for the selected python version
       > The required package might just not (yet) be part of the dependency DB currently used.
       > The DB can be updated by specifying 'pypiDataRev' when importing mach-nix.
       > For examples see: https://github.com/DavHau/mach-nix/blob/master/examples.md
       > If it still doesn't work, there might have bene an error while building the DB.
       > Please open an issue at: https://github.com/DavHau/mach-nix/issues/new
       >
```

eliandoran, 2023-03-11:
@Mastergamer433 , this seems to be caused by using a newer nixpkgs. Seems that there is some issue with the IDF Python dependencies as per #12 .

The only solution I can see at the moment is to pin the `nixpkgs` version to the same as in the [flake.lock](https://github.com/mirrexagon/nixpkgs-esp-dev/blob/master/flake.lock) of `nixpkgs-esp-dev`.

Open your `flake.lock` and change the `nixpkgs`:

```json
    "nixpkgs": {
      "locked": {
        "lastModified": 1636377693,
        "narHash": "sha256-mxnUQZEA361xcupT9RIEcWrNer/+Ik/pbRMDmdRr8gQ=",
        "owner": "nixos",
        "repo": "nixpkgs",
        "rev": "c11d08f02390aab49e7c22e6d0ea9b176394d961",
        "type": "github"
      },
      "original": {
        "owner": "nixos",
        "ref": "nixpkgs-unstable",
        "repo": "nixpkgs",
        "type": "github"
      }
    },
```

Maybe there might be a way to use the `nixpkgs` as defined in the flake of `nixpkgs-esp-dev`, but unfortunately I have no idea how to do that.

Mastergamer433, 2023-03-22:
I'll try that when I get home. Sorry I have not responded. I have stuff to do in school.

mirrexagon, 2023-06-28:
As well as #12, #20 may have broken this use case, since the tools are no longer exported from the flake - let me know if your use case is broken now and I'll see what I can do.

I think @eliandoran's suggestion is the best for now - I know how to set other flakes' nixpkgs to follow the top-level one, but not the other way around.

I'm going to close this for inactivity - if you still have problems, you can reopen it.

closed 2023-06-28 by mirrexagon