mirrexagon/nixpkgs-esp-dev #17: WARNING: directory for tool [...] is present, but tool was not found

eliandoran, 2023-03-11:
Hello,

I am trying to set up a build environment for an ESP32-S3 project, but I was having some warnings.

I managed to reproduce these warnings directly in `nixpkgs-esp-dev` by calling:

```
nix develop .#esp32s3-idf
idf_tools.py list
```

I get the following output:

```
* xtensa-esp32-elf: Toolchain for Xtensa (ESP32) based on GCC
WARNING: directory for tool xtensa-esp32-elf version esp-2021r2-patch3-8.4.0 is present, but tool was not found
  - esp-2021r2-patch3-8.4.0 (recommended)
* xtensa-esp32s2-elf: Toolchain for Xtensa (ESP32-S2) based on GCC
WARNING: directory for tool xtensa-esp32s2-elf version esp-2021r2-patch3-8.4.0 is present, but tool was not found
  - esp-2021r2-patch3-8.4.0 (recommended)
* xtensa-esp32s3-elf: Toolchain for Xtensa (ESP32-S3) based on GCC
WARNING: directory for tool xtensa-esp32s3-elf version esp-2021r2-patch3-8.4.0 is present, but tool was not found
  - esp-2021r2-patch3-8.4.0 (recommended)
* xtensa-clang: LLVM for Xtensa (ESP32, ESP32-S2) based on clang (optional)
  - 12.0.1-d9341b81fc (recommended)
* riscv32-esp-elf: Toolchain for 32-bit RISC-V based on GCC
WARNING: directory for tool riscv32-esp-elf version esp-2021r2-patch3-8.4.0 is present, but tool was not found
  - esp-2021r2-patch3-8.4.0 (recommended)
* esp32ulp-elf: Toolchain for ESP32 ULP coprocessor
WARNING: directory for tool esp32ulp-elf version 2.28.51-esp-20191205 is present, but tool was not found
  - 2.28.51-esp-20191205 (recommended)
* esp32s2ulp-elf: Toolchain for ESP32-S2 and ESP32-S3 ULP coprocessors
WARNING: directory for tool esp32s2ulp-elf version 2.28.51-esp-20191205 is present, but tool was not found
  - 2.28.51-esp-20191205 (recommended)
* cmake: CMake build system (optional)
  - 3.20.3 (recommended)
* openocd-esp32: OpenOCD for ESP32
WARNING: directory for tool openocd-esp32 version v0.11.0-esp32-20211220 is present, but tool was not found
  - v0.11.0-esp32-20211220 (recommended)
* ninja: Ninja build system (optional)
  - 1.10.2 (recommended)
```

Would you happen to know what is causing this warnings? 

During build, these warnings look slightly different:

```
WARNING: directory for tool xtensa-esp32-elf version esp-2021r2-patch3-8.4.0 is present, but tool was not found
ERROR: tool xtensa-esp32-elf has no installed versions. Please run '/nix/store/li7ixzvasab190cj5dywrbab9dlp4zk2-esp-idf-v4.4.1/install.sh' to install it.
WARNING: directory for tool xtensa-esp32s2-elf version esp-2021r2-patch3-8.4.0 is present, but tool was not found
ERROR: tool xtensa-esp32s2-elf has no installed versions. Please run '/nix/store/li7ixzvasab190cj5dywrbab9dlp4zk2-esp-idf-v4.4.1/install.sh' to install it.
WARNING: directory for tool xtensa-esp32s3-elf version esp-2021r2-patch3-8.4.0 is present, but tool was not found
WARNING: directory for tool riscv32-esp-elf version esp-2021r2-patch3-8.4.0 is present, but tool was not found
ERROR: tool riscv32-esp-elf has no installed versions. Please run '/nix/store/li7ixzvasab190cj5dywrbab9dlp4zk2-esp-idf-v4.4.1/install.sh' to install it.
WARNING: directory for tool esp32ulp-elf version 2.28.51-esp-20191205 is present, but tool was not found
ERROR: tool esp32ulp-elf has no installed versions. Please run '/nix/store/li7ixzvasab190cj5dywrbab9dlp4zk2-esp-idf-v4.4.1/install.sh' to install it.
WARNING: directory for tool esp32s2ulp-elf version 2.28.51-esp-20191205 is present, but tool was not found
ERROR: tool esp32s2ulp-elf has no installed versions. Please run '/nix/store/li7ixzvasab190cj5dywrbab9dlp4zk2-esp-idf-v4.4.1/install.sh' to install it.
WARNING: directory for tool openocd-esp32 version v0.11.0-esp32-20211220 is present, but tool was not found
ERROR: tool openocd-esp32 has no installed versions. Please run '/nix/store/li7ixzvasab190cj5dywrbab9dlp4zk2-esp-idf-v4.4.1/install.sh' to install it.
```

Running `install.sh` downloads some packages to the user home directory and fails because it requires `pip`. Shouldn't the install script be run as part of the `nixpkgs-esp-dev` build?

mirrexagon, 2023-05-11:
We can't run `install.sh` in the Nix build environment if it has to download from the network. If the warnings don't stop you from building projects, then I'm not too worried. We might want to figure out if there's a file we can modify to tell ESP-IDF where these tools are when they are installed by Nix, but even then the shell environments for each different ESP32 variant don't install all those tools at the same time.

---

Something this points out though is that the ULP co-processor toolchains are not explicitly packaged - I'm not sure they come with the toolchains for the ESP32 variants that have the ULP co-processor.

mirrexagon, 2023-06-23:
#15 updated ESP-IDF to v5.0.2 and I don't get these warnings - I also haven't run `install.sh` so I'm not sure if that changes things.

It does point of though that I do need to at least update OpenOCD to the recommended version for ESP-IDF v5.0.2, so I will at least do that, and I'll see if I can get the ULP toolchains too.

mirrexagon, 2023-06-28:
#20 makes it so now tools are fetched based on the `tools.json` file in the ESP-IDF checkout - I didn't quite set up all of them, but it should be easy to add ones that aren't currently downloaded and installed. I still don't get the errors/warnings, so I'm going to close this - if you still get the issue, let me know.

closed 2023-06-28 by mirrexagon