mirrexagon/nixpkgs-esp-dev #18: Package ‘python-2.7.18.6’ is marked as insecure, refusing to evaluate.

pkese, 2023-03-17:
Sorry if I'm posting some stupid novice question (I'm not very familiar with nix), but here's what I'm seeing when trying to enter the shell on ubuntu 22.04.

```sh
> nix-shell nixpkgs-esp-dev/shells/esp32-idf.nix 

error: Package ‘python-2.7.18.6’ in /nix/store/b5bg344zj2mbf0cgwnizfngw41wk9795-nixpkgs/nixpkgs/pkgs/development/interpreters/python/cpython/2.7/default.nix:330
is marked as insecure, refusing to evaluate.

       Known issues:
        - Python 2.7 has reached its end of life after 2020-01-01. See https://www.python.org/doc/sunset-python-2/.

       You can install it anyway by allowing this package, using the
       following methods:

       a) To temporarily allow all insecure packages, you can use an environment
          variable for a single invocation of the nix tools:

            $ export NIXPKGS_ALLOW_INSECURE=1

        Note: For `nix shell`, `nix build`, `nix develop` or any other Nix 2.4+
        (Flake) command, `--impure` must be passed in order to read this
        environment variable.

       b) for `nixos-rebuild` you can add ‘python-2.7.18.6’ to
          `nixpkgs.config.permittedInsecurePackages` in the configuration.nix,
          like so:

            {
              nixpkgs.config.permittedInsecurePackages = [
                "python-2.7.18.6"
              ];
            }

       c) For `nix-env`, `nix-build`, `nix-shell` or any other Nix command you can add
          ‘python-2.7.18.6’ to `permittedInsecurePackages` in
          ~/.config/nixpkgs/config.nix, like so:

            {
              permittedInsecurePackages = [
                "python-2.7.18.6"
              ];
            }
(use '--show-trace' to show detailed location information)
```

mirrexagon, 2023-05-11:
Since Python 2.x is now end-of-life, nixpkgs has marked it as insecure.

You might still be able to get a working shell if you have Nix flake support with `nix develop .#esp32-idf`, but that uses a pretty old commit of nixpkgs and I don't know if prebuilt packages are still available from `cache.nixos.org`.

Also see #10 and #12.

mirrexagon, 2023-06-28:
I'm not actually sure why we would be pulling in Python 2.7 - we should be using Python 3.x for everything as far as I know.

I'd hope that the solution to #12 would make this irrelevant, but since I don't understand why Python 2.7 is even involved, I will leave this open.

mirrexagon, 2023-06-30:
I made updates to how the Python environment works in #23 - hopefully that fix also fixes this (since you were using your system nixpkgs and not using `nixpkgs-esp-dev` via the flake, and this update should make newer nixpkgs work with `nixpkgs-esp-dev` again.

Let me know if you run into this again.

closed 2023-06-30 by mirrexagon