mirrexagon/nixpkgs-esp-dev #22: [Errno 71] Protocol error when trying to run idf.py for ESP32-S2

loco-choco, 2023-06-29:
I am having issues trying to use idf.py with my esp32-s2, where the program crashes with the error `OSError: [Errno 71] Protocol error`.
Here is the full error stack of running `idf.py monitor`:
```shell
Executing action: monitor
Serial port /dev/ttyACM0
Connecting...
Detecting chip type... Unsupported detection protocol, switching and trying again...
Detecting chip type... ESP32-S2
Running idf_monitor in directory /home/locochoco/semear/esp32/esp-idf-template
Executing "/nix/store/8dxxjbiyxwkvh53q5kh6nydla2anacgi-python3-3.9.6/bin/python /nix/store/yb9bwl07rrjrav4wr2qr82zgvh4r9615-esp-idf-v5.0.2/tools/idf_monitor.py -p /dev/ttyACM0 -b 115200 --toolchain-prefix xtensa-esp32s2-elf- --target esp32s2 /home/locochoco/semear/esp32/esp-idf-template/build/app-template.elf -m '/nix/store/8dxxjbiyxwkvh53q5kh6nydla2anacgi-python3-3.9.6/bin/python' '/nix/store/yb9bwl07rrjrav4wr2qr82zgvh4r9615-esp-idf-v5.0.2/tools/idf.py'"...
--- idf_monitor on /dev/ttyACM0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/nix/store/8dxxjbiyxwkvh53q5kh6nydla2anacgi-python3-3.9.6/lib/python3.9/threading.py", line 973, in _bootstrap_inner
    self.run()
  File "/nix/store/8dxxjbiyxwkvh53q5kh6nydla2anacgi-python3-3.9.6/lib/python3.9/threading.py", line 910, in run
    self._target(*self._args, **self._kwargs)
  File "/nix/store/yb9bwl07rrjrav4wr2qr82zgvh4r9615-esp-idf-v5.0.2/tools/idf_monitor_base/stoppable_thread.py", line 47, in _run_outer
    self.run()
  File "/nix/store/yb9bwl07rrjrav4wr2qr82zgvh4r9615-esp-idf-v5.0.2/tools/idf_monitor_base/serial_reader.py", line 55, in run
    self.serial.rts = low      # Set rts/dtr to the reset state
  File "/nix/store/yb9bwl07rrjrav4wr2qr82zgvh4r9615-esp-idf-v5.0.2/lib/python3.9/site-packages/serial/serialutil.py", line 463, in rts
    self._update_rts_state()
  File "/nix/store/yb9bwl07rrjrav4wr2qr82zgvh4r9615-esp-idf-v5.0.2/lib/python3.9/site-packages/serial/serialposix.py", line 706, in _update_rts_state
    fcntl.ioctl(self.fd, TIOCMBIS, TIOCM_RTS_str)
OSError: [Errno 71] Protocol error
```
and here is the dmesg of connecting and running the software, I see nothing that could point to the source of the issue
```shell
[ 3545.654100] usb 1-2: new full-speed USB device number 11 using xhci_hcd
[ 3545.781790] usb 1-2: New USB device found, idVendor=303a, idProduct=0002, bcdDevice= 7.23
[ 3545.781802] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 3545.781808] usb 1-2: Product: ESP32-S2
[ 3545.781812] usb 1-2: Manufacturer: Espressif
[ 3545.781816] usb 1-2: SerialNumber: 0
[ 3545.784573] cdc_acm 1-2:1.0: ttyACM0: USB ACM device
```
I am running the tool stack from `nix develop github:mirrexagon/nixpkgs-esp-dev#esp32s2-idf`. I am not sure if this would be the best place to make an issue about this, but I saw that the flake.lock is one years old, so I am not sure if the issue could be related to that.

mirrexagon, 2023-06-30:
Are you using the native USB port on the ESP32-S2, and not the UART port (via a USB-to-serial chip)? The seems to be implied by the Product being ESP32-S2 and the manufacturer being Espressif. For reference, the UART port on my ESP32-S2-DevKitC-1 comes up as:

```
[38305.642884] usb 3-2: new full-speed USB device number 57 using xhci_hcd
[38305.771032] usb 3-2: New USB device found, idVendor=10c4, idProduct=ea60, bcdDevice= 1.00
[38305.771041] usb 3-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[38305.771044] usb 3-2: Product: CP2102N USB to UART Bridge Controller
[38305.771046] usb 3-2: Manufacturer: Silicon Labs
[38305.771048] usb 3-2: SerialNumber: 861d6d672424ed119db58ae8f9a97352
[38305.773299] cp210x 3-2:1.0: cp210x converter detected
[38305.775060] usb 3-2: cp210x converter now attached to ttyUSB0
```

What program are you running on the ESP32-S2? I wonder if the virtual serial port from the native USB controller doesn't support toggling RTS (which I think `idf.py monitor` is trying to do to reset the board).

I've also updated `flake.lock` FINALLY in #23, so you could also try again and see if that helps.

loco-choco, 2023-06-30:
I am running the hello world template from https://github.com/espressif/esp-idf-template, something I should add is that my esp32s2 is an esp32s2 mini, and yes I am using the native usb port. Strangely is that I could get the arduino ide in windows to work with it, so the tools should be there.

loco-choco, 2023-06-30:
Strange, now after having a program loaded it outputs a different error:
```shell
idf.py monitor
Executing action: monitor
Serial port /dev/ttyACM0
Connecting.............
/dev/ttyACM0 failed to connect: write failed: [Errno 19] No such device
No serial ports found. Connect a device, or use '-p PORT' option to set a specific port.
```
```shell
[ 1298.043775] usb 1-14: new full-speed USB device number 13 using xhci_hcd
[ 1298.171868] usb 1-14: New USB device found, idVendor=303a, idProduct=8141, bcdDevice= 1.00
[ 1298.171880] usb 1-14: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1298.171886] usb 1-14: Product: DENEYAP MINI
[ 1298.171891] usb 1-14: Manufacturer: Turkish Technnology Team Foundation (T3)
[ 1298.171895] usb 1-14: SerialNumber: 48:27:e2:5d:f3:46
[ 1298.175056] cdc_acm 1-14:1.0: ttyACM0: USB ACM device
<---- moment I run idf.py monitor ---->
[ 1373.499395] usb 1-14: USB disconnect, device number 13
[ 1373.499519] cdc_acm 1-14:1.1: acm_start_wb - usb_submit_urb(write bulk) failed: -19
[ 1373.751671] usb 1-14: new full-speed USB device number 14 using xhci_hcd
[ 1373.879413] usb 1-14: New USB device found, idVendor=303a, idProduct=0002, bcdDevice= 7.23
[ 1373.879426] usb 1-14: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1373.879432] usb 1-14: Product: ESP32-S2
[ 1373.879436] usb 1-14: Manufacturer: Espressif
[ 1373.879440] usb 1-14: SerialNumber: 0
[ 1373.882076] cdc_acm 1-14:1.0: ttyACM0: USB ACM device
```
And if I run the program again (without disconnecting the esp or rebooting it) the same issue persists
```shell
idf.py monitor
Executing action: monitor
Serial port /dev/ttyACM0
Connecting...
Detecting chip type... Unsupported detection protocol, switching and trying again...
Detecting chip type... ESP32-S2
Running idf_monitor in directory /home/locochoco/semear/esp32/esp-idf-template
Executing "/nix/store/7zgswij5795q5k9wkl33fk638b98pzbl-python3-3.10.12-env/bin/python3.10 /nix/store/10am01kfgmsidrw1jfnwbpqlvwhmsqn2-esp-idf-v5.0.2/tools/idf_monitor.py -p /dev/ttyACM0 -b 115200 --toolchain-prefix xtensa-esp32s2-elf- --target esp32s2 /home/locochoco/semear/esp32/esp-idf-template/build/app-template.elf -m '/nix/store/7zgswij5795q5k9wkl33fk638b98pzbl-python3-3.10.12-env/bin/python3.10' '/nix/store/10am01kfgmsidrw1jfnwbpqlvwhmsqn2-esp-idf-v5.0.2/tools/idf.py'"...
--- idf_monitor on /dev/ttyACM0 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
Exception in thread Thread-2 (_run_outer):
Traceback (most recent call last):
  File "/nix/store/1r6n7v2wam7gkr18gxccpg7p5ywgw551-python3-3.10.12/lib/python3.10/threading.py", line 1016, in _bootstrap_inner
    self.run()
  File "/nix/store/1r6n7v2wam7gkr18gxccpg7p5ywgw551-python3-3.10.12/lib/python3.10/threading.py", line 953, in run
    self._target(*self._args, **self._kwargs)
  File "/nix/store/10am01kfgmsidrw1jfnwbpqlvwhmsqn2-esp-idf-v5.0.2/tools/idf_monitor_base/stoppable_thread.py", line 47, in _run_outer
    self.run()
  File "/nix/store/10am01kfgmsidrw1jfnwbpqlvwhmsqn2-esp-idf-v5.0.2/tools/idf_monitor_base/serial_reader.py", line 55, in run
    self.serial.rts = low      # Set rts/dtr to the reset state
  File "/nix/store/7zgswij5795q5k9wkl33fk638b98pzbl-python3-3.10.12-env/lib/python3.10/site-packages/serial/serialutil.py", line 463, in rts
    self._update_rts_state()
  File "/nix/store/7zgswij5795q5k9wkl33fk638b98pzbl-python3-3.10.12-env/lib/python3.10/site-packages/serial/serialposix.py", line 706, in _update_rts_state
    fcntl.ioctl(self.fd, TIOCMBIS, TIOCM_RTS_str)
OSError: [Errno 71] Protocol error

```

```shell
[ 1687.562540] cdc_acm 1-14:1.1: acm_start_wb - usb_submit_urb(write bulk) failed: -19
[ 1687.818619] usb 1-14: new full-speed USB device number 20 using xhci_hcd
[ 1687.946198] usb 1-14: New USB device found, idVendor=303a, idProduct=0002, bcdDevice= 7.23
[ 1687.946212] usb 1-14: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1687.946218] usb 1-14: Product: ESP32-S2
[ 1687.946222] usb 1-14: Manufacturer: Espressif
[ 1687.946226] usb 1-14: SerialNumber: 0
[ 1687.949163] cdc_acm 1-14:1.0: ttyACM0: USB ACM device
<---- moment I run again idf.py monitor ---->
[ 1726.487086] usb 1-14: USB disconnect, device number 20
[ 1726.740532] usb 1-14: new full-speed USB device number 21 using xhci_hcd
[ 1726.868439] usb 1-14: New USB device found, idVendor=303a, idProduct=8141, bcdDevice= 1.00
[ 1726.868451] usb 1-14: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 1726.868457] usb 1-14: Product: DENEYAP MINI
[ 1726.868462] usb 1-14: Manufacturer: Turkish Technnology Team Foundation (T3)
[ 1726.868466] usb 1-14: SerialNumber: 48:27:e2:5d:f3:46
[ 1726.871370] cdc_acm 1-14:1.0: ttyACM0: USB ACM device
```

loco-choco, 2023-07-01:
Another error dump from flashing, different code erroring out but same error, maybe this can give us some insight on what is happening.
```shell
idf.py flash
Executing action: flash
Serial port /dev/ttyACM0
Connecting...
Detecting chip type... Unsupported detection protocol, switching and trying again...
Detecting chip type... ESP32-S2
Running ninja in directory /home/locochoco/semear/esp32/esp-idf-template/build
Executing "ninja flash"...
[1/5] cd /home/locochoco/semear/esp32/esp-idf-template/build/esp-idf/esptool_py && /nix/store/7zgswij5795q5k9wkl33fk638b98pzbl-python3-3.10.12-env/bin/python3.10 /nix/store/10am01kfgmsidrw1jfnwbpqlvwhmsqn2-esp-idf-v5.0.2/components/partition_table/check_sizes.py --offset 0x8000 partition --type app /home/locochoco/semear/esp32/esp-idf-template/build/partition_table/partition-table.bin /home/locochoco/semear/esp32/esp-idf-template/build/app-template.bin
app-template.bin binary size 0x27990 bytes. Smallest app partition is 0x100000 bytes. 0xd8670 bytes (85%) free.
[2/5] Performing build step for 'bootloader'
[1/1] cd /home/locochoco/semear/esp32/esp-idf-template/build/bootloader/esp-idf/esptool_py && /nix/store/7zgswij5795q5k9wkl33fk638b98pzbl-python3-3.10.12-env/bin/python3.10 /nix/store/10am01kfgmsidrw1jfnwbpqlvwhmsqn2-esp-idf-v5.0.2/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x1000 /home/locochoco/semear/esp32/esp-idf-template/build/bootloader/bootloader.bin
Bootloader binary size 0x5430 bytes. 0x1bd0 bytes (25%) free.
[3/5] No install step for 'bootloader'
[4/5] Completed 'bootloader'
[4/5] cd /nix/store/10am01kfgmsidrw1jfnwbpqlvwhmsqn2-esp-idf-v5.0.2/components/esptool_py && /nix/store/f8inixjb36rcxvjwsghqlqqxgbrdrvly-cmake-3.26.4/bin/cmake -D IDF_PATH=/nix/store/10am01kfgmsidrw1jfnwbpqlvwhmsqn2-esp-idf-v5.0.2 -D "SERIAL_TOOL=/nix/store/7zgswij5795q5k9wkl33fk638b98pzbl-python3-3.10.12-env/bin/python3.10;;/nix/store/10am01kfgmsidrw1jfnwbpqlvwhmsqn2-esp-idf-v5.0.2/components/esptool_py/esptool/esptool.py;--chip;esp32s2" -D "SERIAL_TOOL_ARGS=--before=default_reset;--after=hard_reset;write_flash;@flash_args" -D WORKING_DIRECTORY=/home/locochoco/semear/esp32/esp-idf-template/build -P /nix/store/10am01kfgmsidrw1jfnwbpqlvwhmsqn2-esp-idf-v5.0.2/components/esptool_py/run_serial_tool.cmake
esptool esp32s2 -p /dev/ttyACM0 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size 2MB 0x1000 bootloader/bootloader.bin 0x10000 app-template.bin 0x8000 partition_table/partition-table.bin
esptool.py v4.5.1
Serial port /dev/ttyACM0
Traceback (most recent call last):
  File "/nix/store/1r6n7v2wam7gkr18gxccpg7p5ywgw551-python3-3.10.12/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/nix/store/1r6n7v2wam7gkr18gxccpg7p5ywgw551-python3-3.10.12/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/nix/store/7zgswij5795q5k9wkl33fk638b98pzbl-python3-3.10.12-env/lib/python3.10/site-packages/esptool/__main__.py", line 9, in <module>
    esptool._main()
  File "/nix/store/7zgswij5795q5k9wkl33fk638b98pzbl-python3-3.10.12-env/lib/python3.10/site-packages/esptool/__init__.py", line 1032, in _main
    main()
  File "/nix/store/7zgswij5795q5k9wkl33fk638b98pzbl-python3-3.10.12-env/lib/python3.10/site-packages/esptool/__init__.py", line 674, in main
    esp = esp or get_default_connected_device(
  File "/nix/store/7zgswij5795q5k9wkl33fk638b98pzbl-python3-3.10.12-env/lib/python3.10/site-packages/esptool/__init__.py", line 921, in get_default_connected_device
    _esp.connect(before, connect_attempts)
  File "/nix/store/7zgswij5795q5k9wkl33fk638b98pzbl-python3-3.10.12-env/lib/python3.10/site-packages/esptool/loader.py", line 640, in connect
    last_error = self._connect_attempt(reset_strategy, mode)
  File "/nix/store/7zgswij5795q5k9wkl33fk638b98pzbl-python3-3.10.12-env/lib/python3.10/site-packages/esptool/loader.py", line 525, in _connect_attempt
    reset_strategy()  # Reset the chip to bootloader (download mode)
  File "/nix/store/7zgswij5795q5k9wkl33fk638b98pzbl-python3-3.10.12-env/lib/python3.10/site-packages/esptool/reset.py", line 85, in __call__
    self._setDTRandRTS(False, True)  # IO0=HIGH & EN=LOW, chip in reset
  File "/nix/store/7zgswij5795q5k9wkl33fk638b98pzbl-python3-3.10.12-env/lib/python3.10/site-packages/esptool/reset.py", line 58, in _setDTRandRTS
    fcntl.ioctl(self.port.fileno(), TIOCMSET, struct.pack("I", status))
OSError: [Errno 71] Protocol error
Connecting...
CMake Error at run_serial_tool.cmake:55 (message):
  
  /nix/store/7zgswij5795q5k9wkl33fk638b98pzbl-python3-3.10.12-env/bin/python3.10;;/nix/store/10am01kfgmsidrw1jfnwbpqlvwhmsqn2-esp-idf-v5.0.2/components/esptool_py/esptool/esptool.py;--chip;esp32s2
  failed

FAILED: CMakeFiles/flash /home/locochoco/semear/esp32/esp-idf-template/build/CMakeFiles/flash 
cd /nix/store/10am01kfgmsidrw1jfnwbpqlvwhmsqn2-esp-idf-v5.0.2/components/esptool_py && /nix/store/f8inixjb36rcxvjwsghqlqqxgbrdrvly-cmake-3.26.4/bin/cmake -D IDF_PATH=/nix/store/10am01kfgmsidrw1jfnwbpqlvwhmsqn2-esp-idf-v5.0.2 -D "SERIAL_TOOL=/nix/store/7zgswij5795q5k9wkl33fk638b98pzbl-python3-3.10.12-env/bin/python3.10;;/nix/store/10am01kfgmsidrw1jfnwbpqlvwhmsqn2-esp-idf-v5.0.2/components/esptool_py/esptool/esptool.py;--chip;esp32s2" -D "SERIAL_TOOL_ARGS=--before=default_reset;--after=hard_reset;write_flash;@flash_args" -D WORKING_DIRECTORY=/home/locochoco/semear/esp32/esp-idf-template/build -P /nix/store/10am01kfgmsidrw1jfnwbpqlvwhmsqn2-esp-idf-v5.0.2/components/esptool_py/run_serial_tool.cmake
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /home/locochoco/semear/esp32/esp-idf-template/build/log/idf_py_stderr_output_50881 and /home/locochoco/semear/esp32/esp-idf-template/build/log/idf_py_stdout_output_50881

```
 Perhaps this is some issue with the way python is being packaged?

loco-choco, 2023-07-04:
Figured out the issue was with needing to use `dfu-util` to allow the board to be flashed with usb.

closed 2023-07-04 by loco-choco