mirrexagon/nixpkgs-esp-dev #30: Can't execute openocd using the flake

tyalie, 2023-10-22:
When I try to use the openocd bundled in the esp32-idf flake, I get the following error during execution

```
/nix/store/6zyhvqamxp5j6fqy2nx1m879klxninq7-openocd-esp32-esp-idf-v5.1/bin/openocd-unwrapped: error while loading shared libraries: libudev.so.1: cannot open shared object file: No such file or directory
```

Looking with `ldd` I can see that it has multiple unmet dependencies. 
```
ldd /nix/store/6zyhvqamxp5j6fqy2nx1m879klxninq7-openocd-esp32-esp-idf-v5.1/bin/openocd-unwrapped
	linux-vdso.so.1 (0x00007ffff7fc8000)
	libudev.so.1 => not found
	libusb-1.0.so.0 => not found
	libm.so.6 => /nix/store/3n58xw4373jp0ljirf06d8077j15pc4j-glibc-2.37-8/lib/libm.so.6 (0x00007ffff7ee2000)
	libz.so.1 => not found
	libutil.so.1 => /nix/store/3n58xw4373jp0ljirf06d8077j15pc4j-glibc-2.37-8/lib/libutil.so.1 (0x00007ffff7edd000)
	libdl.so.2 => /nix/store/3n58xw4373jp0ljirf06d8077j15pc4j-glibc-2.37-8/lib/libdl.so.2 (0x00007ffff7ed6000)
	libc.so.6 => /nix/store/3n58xw4373jp0ljirf06d8077j15pc4j-glibc-2.37-8/lib/libc.so.6 (0x00007ffff7cf0000)
	/lib64/ld-linux-x86-64.so.2 => /nix/store/3n58xw4373jp0ljirf06d8077j15pc4j-glibc-2.37-8/lib64/ld-linux-x86-64.so.2 (0x00007ffff7fca000)
```

I'm totally new to nix, so I'm a bit baffled what the right way forward would be

mirrexagon, 2024-03-17:
It seems that even though those dependencies are specificied in the `targetPkgs` for the FHSUserEnv used to run those tools, they don't get picked up. I'm not sure why they don't get picked up.

BastienGermond, 2024-03-27:
This got patched by #44.

mirrexagon, 2024-03-29:
Whoops, I missed closing this, thanks.

closed 2024-03-29 by mirrexagon