mirrexagon/nixpkgs-esp-dev #45: Broken C compiler

liamdiprose, 2024-03-22:
Version: Current HEAD of `dev-esp32` (10d0a26f)

The build fails when checking for a working C compiler:

```
-- Check for working C compiler: /nix/store/bbs40g5kz34lc0aj0x8da1d8h7kpak4k-xtensa-esp32-elf-esp-idf-v5.1.2/bin/xtensa-esp32-elf-gcc - broken
```

The most likely cause in the error trace:
```

<details>
<summary>Full Error Trace</summary>

```
liam@delson ~/p/nodemcu-firmware (dev-esp32)> make
/nix/store/859hchx1m5vddd17xlijrbjvjqmw4yrk-esp-idf-v5.1.2/tools/idf.py  "all"
Executing action: all (aliases: build)
Running cmake in directory /home/liam/projects/nodemcu-firmware/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=/nix/store/41253d4jjd23jjdaiidgnb8q2qambh1a-python3-3.11.7-env/bin/python3.11 -DESP_PLATFORM=1 -DCCACHE_ENABLE=0 /home/liam/projects/nodemcu-firmware"...
-- IDF_TARGET not set, using default target: esp32
-- Found Git: /nix/store/za04kin80ybavb7qp577dghpax4mf82z-git-2.43.0/bin/git (found version "2.43.0")
-- git rev-parse returned 'fatal: not a git repository (or any of the parent directories): .git'
fatal: not a git repository (or any of the parent directories): .git
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
-- The ASM compiler identification is unknown
-- Found assembler: /nix/store/bbs40g5kz34lc0aj0x8da1d8h7kpak4k-xtensa-esp32-elf-esp-idf-v5.1.2/bin/xtensa-esp32-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - failed
-- Check for working C compiler: /nix/store/bbs40g5kz34lc0aj0x8da1d8h7kpak4k-xtensa-esp32-elf-esp-idf-v5.1.2/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /nix/store/bbs40g5kz34lc0aj0x8da1d8h7kpak4k-xtensa-esp32-elf-esp-idf-v5.1.2/bin/xtensa-esp32-elf-gcc - broken
CMake Error at /nix/store/jacf2kn4dfj99c3ywbvfyg6w53xvxsfl-cmake-3.27.8/share/cmake-3.27/Modules/CMakeTestCCompiler.cmake:67 (message):
  The C compiler

    "/nix/store/bbs40g5kz34lc0aj0x8da1d8h7kpak4k-xtensa-esp32-elf-esp-idf-v5.1.2/bin/xtensa-esp32-elf-gcc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: '/home/liam/projects/nodemcu-firmware/build/CMakeFiles/CMakeScratch/TryCompile-BJ3aU4'

    Run Build Command(s): /nix/store/5lbxsj5mnz95rq5hkq7ixxb1cg96k07g-ninja-1.11.1/bin/ninja -v cmTC_87cfe
    [1/2] /nix/store/bbs40g5kz34lc0aj0x8da1d8h7kpak4k-xtensa-esp32-elf-esp-idf-v5.1.2/bin/xtensa-esp32-elf-gcc   -mlongcalls -Wno-frame-address -o CMakeFiles/cmTC_87cfe.dir/testCCompiler.c.obj -c /home/liam/projects/nodemcu-firmware/build/CMakeFiles/CMakeScratch/TryCompile-BJ3aU4/testCCompiler.c
    FAILED: CMakeFiles/cmTC_87cfe.dir/testCCompiler.c.obj
    /nix/store/bbs40g5kz34lc0aj0x8da1d8h7kpak4k-xtensa-esp32-elf-esp-idf-v5.1.2/bin/xtensa-esp32-elf-gcc   -mlongcalls -Wno-frame-address -o CMakeFiles/cmTC_87cfe.dir/testCCompiler.c.obj -c /home/liam/projects/nodemcu-firmware/build/CMakeFiles/CMakeScratch/TryCompile-BJ3aU4/testCCompiler.c
    /nix/store/bbs40g5kz34lc0aj0x8da1d8h7kpak4k-xtensa-esp32-elf-esp-idf-v5.1.2/bin/xtensa-esp32-elf-gcc: line 2: /nix/store/bbs40g5kz34lc0aj0x8da1d8h7kpak4k-xtensa-esp32-elf-esp-idf-v5.1.2/bin/.xtensa-esp32-elf-gcc-wrapped: cannot execute: required file not found
    ninja: build stopped: subcommand failed.

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  /nix/store/859hchx1m5vddd17xlijrbjvjqmw4yrk-esp-idf-v5.1.2/tools/cmake/project.cmake:448 (__project)
  CMakeLists.txt:10 (project)

-- Configuring incomplete, errors occurred!
cmake failed with exit code 1, output of the command is in the /home/liam/projects/nodemcu-firmware/build/log/idf_py_stderr_output_1468265 and /home/liam/projects/nodemcu-firmware/build/log/idf_py_stdout_output_1468265
make: *** [Makefile:18: all] Error 2
```
</details>

liamdiprose, 2024-03-22:
Issue is resolved by #42

mirrexagon, 2024-03-23:
I merged #42, let me know if there's still problems.

closed 2024-03-23 by mirrexagon