mirrexagon/nixpkgs-esp-dev #51: esp_rfc2217_server.py not working because of a syntax error in python3.11-esptool-4.6.2/bin/esptool.py

mksafavi, 2024-05-23:
I noticed that `esp_rfc2217_server.py` doesn't work. I'm not sure if this is the only command that has this issue.
Let me know if there's any other info that I should share.

command to run:
```
esp_rfc2217_server.py -p 4400 /dev/ttyACM0
```
error:
```
Traceback (most recent call last):
  File "/nix/store/wn78z9b8lijylzy8znzmdj021z54iz6f-python3.11-esptool-4.6.2/bin/.esp_rfc2217_server.py-wrapped", line 40, in <module>
    from esptool.config import load_config_file
  File "/nix/store/wn78z9b8lijylzy8znzmdj021z54iz6f-python3.11-esptool-4.6.2/bin/esptool.py", line 2
    PATH=${PATH:+':'$PATH':'}
         ^
SyntaxError: invalid syntax
```

expected output:
```
INFO:root:RFC 2217 TCP/IP to Serial redirector - type Ctrl-C / BREAK to quit
INFO:root:Serving serial port: /dev/ttyACM0
INFO:root:TCP/IP port: 4400

```

danielemery, 2024-06-18:
Just chiming in to say I am having the same issue

mksafavi, 2024-07-03:
I think I found the problem.
when you run `esp_rfc2217_server.py`  it starts `/nix/store/h5kainwxnc8slhhmd9kp2h82srjy0hyr-python3-3.11.9-env/bin/esp_rfc2217_server.py` which is a wrapper around the python source that's here: `/nix/store/h5kainwxnc8slhhmd9kp2h82srjy0hyr-python3-3.11.9-env/bin/.esp_rfc2217_server.py-wrapped`.

in line 40 of `.esp_rfc2217_server.py-wrapped` it tries to import `esptool.py`:
```
from esptool.config import load_config_file
from esptool.reset import ClassicReset, CustomReset, DEFAULT_RESET_DELAY, UnixTightReset
```
but `esptool.py` is also a wrapper. the actual module is here `.esptool.py-wrapped`

I'm not sure if this is a byproduct of building esptool with `buildPythonPackage` instead of `buildPythonApplication`.
But I'll try patching the `.esp_rfc2217_server.py-wrapped` imports to see if it works.

closed 2024-08-22 by mirrexagon