mirrexagon/nixpkgs-esp-dev #53: Unable to create new project

srounce, 2024-06-08:
When running `idf.py create-project` I get the error below. I presume this is because the project files are being copied from the store with the read-only permissions being retained.

```
$ idf.py create-project esp-test
Executing action: create-project
Traceback (most recent call last):
  File "/nix/store/qiqg11z64gm0ahnpa9l13wz7d5xxkij4-esp-idf-v5.1.3/tools/idf.py", line 779, in <module>
    main()
  File "/nix/store/qiqg11z64gm0ahnpa9l13wz7d5xxkij4-esp-idf-v5.1.3/tools/idf.py", line 714, in main
    cli(sys.argv[1:], prog_name=PROG, complete_var=SHELL_COMPLETE_VAR)
  File "/nix/store/41253d4jjd23jjdaiidgnb8q2qambh1a-python3-3.11.7-env/lib/python3.11/site-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/nix/store/41253d4jjd23jjdaiidgnb8q2qambh1a-python3-3.11.7-env/lib/python3.11/site-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "/nix/store/41253d4jjd23jjdaiidgnb8q2qambh1a-python3-3.11.7-env/lib/python3.11/site-packages/click/core.py", line 1720, in invoke
    return _process_result(rv)
           ^^^^^^^^^^^^^^^^^^^
  File "/nix/store/41253d4jjd23jjdaiidgnb8q2qambh1a-python3-3.11.7-env/lib/python3.11/site-packages/click/core.py", line 1657, in _process_result
    value = ctx.invoke(self._result_callback, value, **ctx.params)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/nix/store/41253d4jjd23jjdaiidgnb8q2qambh1a-python3-3.11.7-env/lib/python3.11/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/nix/store/qiqg11z64gm0ahnpa9l13wz7d5xxkij4-esp-idf-v5.1.3/tools/idf.py", line 611, in execute_tasks
    task(ctx, global_args, task.action_args)
  File "/nix/store/qiqg11z64gm0ahnpa9l13wz7d5xxkij4-esp-idf-v5.1.3/tools/idf.py", line 193, in __call__
    self.callback(self.name, context, global_args, **action_args)
  File "/nix/store/qiqg11z64gm0ahnpa9l13wz7d5xxkij4-esp-idf-v5.1.3/tools/idf_py_actions/create_ext.py", line 68, in create_new
    func_action_map[action](target_path, action_args['name'])
  File "/nix/store/qiqg11z64gm0ahnpa9l13wz7d5xxkij4-esp-idf-v5.1.3/tools/idf_py_actions/create_ext.py", line 46, in create_project
    replace_in_file(os.path.join(main_folder, 'CMakeLists.txt'), 'main', name)
  File "/nix/store/qiqg11z64gm0ahnpa9l13wz7d5xxkij4-esp-idf-v5.1.3/tools/idf_py_actions/create_ext.py", line 20, in replace_in_file
    with open(filename, 'r+') as f:
         ^^^^^^^^^^^^^^^^^^^^
PermissionError: [Errno 13] Permission denied: '/home/s/repos/esp-test/main/CMakeLists.txt'
```

mirrexagon, 2024-06-19:
See if #54 fixed this.

mirrexagon, 2024-10-23:
This should be fixed upstream, we are using ESP-IDF v5.3 now.