mirrexagon/nixpkgs-esp-dev #58: parttool.py is not available on bin path

mksafavi, 2024-07-22:
Some of the idf tools are inside `/components` directories and they're symlinked to /bin/
maybe we should add them.

on esp-idf docker image:
```
$ whereis parttool
parttool: /opt/esp/idf/components/partition_table/parttool.py
$ whereis parttool.py
parttool: /opt/esp/idf/components/partition_table/parttool.py
```
they can be assessed like this on nixpkgs-esp-dev:
```
$IDF_PATH/components/partition_table/parttool.py
$IDF_PATH/components/spiffs/spiffsgen.py
```

mirrexagon, 2024-10-23:
Do you know how the Docker container has this set up? I'm looking in the Dockerfile and `idf_tools.py` to try and figure it out so we can do it the same way, but I haven't seen anything so far.

mirrexagon, 2024-10-23:
I found that `export.sh` does this:

```sh
    IDF_ADD_PATHS_EXTRAS="${IDF_PATH}/components/espcoredump"
    IDF_ADD_PATHS_EXTRAS="${IDF_ADD_PATHS_EXTRAS}:${IDF_PATH}/components/partition_table"
    IDF_ADD_PATHS_EXTRAS="${IDF_ADD_PATHS_EXTRAS}:${IDF_PATH}/components/app_update"
```

So I will do that in the ESP-IDF setup hook.

commit 3f748af7 (2024-10-23): Add extra PATH entries from export.sh

Fixes #58
closed 2024-10-23 by mirrexagon