mirrexagon/nixpkgs-esp-dev #62: What to set `ESP_ROM_ELF_DIR`

jr1221, 2024-08-27:
Hi,

`ESP_ROM_ELF_DIR` is not set, which results in failed `gdb` commands.

Can you please advise what to set it to?

jr1221, 2024-08-27:
I got good results with this `export ESP_ROM_ELF_DIR=/nix/store/16d6gy5clh0ri2z5wkcnzdj3l1cbmadq-xtensa-esp-elf-gdb-esp-idf-v5.3`

mirrexagon, 2024-10-23:
According to ESP-IDF v5.3.1's `tools.json`, it should be set to the `esp-rom-elfs` directory. I assume this is to look up debugging information for the burned-in boot ROM if you ever step into it. I will have to install that as a tool, and get exported vars working at the `idf.py` level (as in #61).

quentinmit, 2024-12-07:
I've attempted to address this with #71

mirrexagon, 2025-01-29:
Should be fixed by #71.