mirrexagon/nixpkgs-esp-dev #65: ESP-IDF for vscode cannot find /tools/idf.py in the nix store

Azeirah, 2024-09-22:
[image]

I'm trying to set-up vscodium with the ESP-IDF plugin to do esp development, but as you can see it's not quite working.

I think the issue might have to do with permissions?

Anyone have an idea how I can get the ESP-IDF plugin for VSCode(ium) to work on Nix?

Azeirah, 2024-09-22:
Oh and for reference, I made a flake that combines this project with the vscodium flake. I have to be honest, I let claude do most of it. I'm still learning Nix. For what it's worth, it does work. It gives you a vscodium installation with ESP-IDF pre-installed, so it's practically a portable development environment nixified.

If anyone is interested in trying (or roasting the flake, I'm sorry!) feel free to comment :)

```
{
  description = "ESP-IDF development environment with VSCodium";

  inputs = {
    nixpkgs.url = "nixpkgs/nixpkgs-unstable";
    flake-utils.url = "github:numtide/flake-utils";
    nix-vscode-extensions.url = "github:nix-community/nix-vscode-extensions";
    esp-dev.url = "github:mirrexagon/nixpkgs-esp-dev";
  };

  outputs = { self, nixpkgs, flake-utils, nix-vscode-extensions, esp-dev }:
    flake-utils.lib.eachDefaultSystem (system:
      let
        pkgs = import nixpkgs {
          inherit system;
          overlays = [
            esp-dev.overlays.default
            nix-vscode-extensions.overlays.default
          ];
        };

        extensions = pkgs.vscode-marketplace;

        vscodium-with-extensions = pkgs.vscode-with-extensions.override {
          vscode = pkgs.vscodium;
          vscodeExtensions = [
            extensions.golang.go
            extensions.rust-lang.rust-analyzer
	     extensions.espressif.esp-idf-extension
          ];
        };

      in {
        packages = {
          default = vscodium-with-extensions;
          inherit (pkgs) esp-idf-full;
        };

        devShells.default = pkgs.mkShell {
          buildInputs = [
            vscodium-with-extensions
            pkgs.esp-idf-full
          ];

          shellHook = ''
            alias i='idf.py'

            printf "\033[1;36m"  # Set text color to cyan
            cat << EOF
            ╔════════════════════════════════════════════════════════════════╗
            ║         Welcome to the ESP-IDF Development Environment!        ║
            ╚════════════════════════════════════════════════════════════════╝

            This environment provides you with:

            1. VSCodium with pre-configured extensions for ESP-IDF development
               - Run 'codium' to start the IDE

            2. ESP-IDF tools, including idf.py (aliased to 'i')
               - Run 'i' or 'idf.py' to use the ESP-IDF build system

            To get started with ESP-IDF, visit:
            https://docs.espressif.com/projects/esp-idf/en/stable/esp32/get-started/index.html

            Here's a quick overview of what's available:
            EOF
            printf "\033[0m"  # Reset text color

            printf "\033[1;33m"  # Set text color to yellow
            printf "\nVSCodium with extensions:\n"
            codium --list-extensions
            printf "\033[0m"  # Reset text color

            printf "\033[1;32m"  # Set text color to green
            printf "\nESP-IDF environment:\n"
            i --version
            printf "\033[0m"  # Reset text color

            printf "\n\033[1mTip:\033[0m You can use 'i' as a shortcut for 'idf.py'. For example:\n"
            printf "  i build\n"
            printf "  i flash\n"
            printf "  i monitor\n"

            printf "\n\033[1mHappy coding with ESP-IDF!\033[0m\n"
          '';
	};
      }
    );
}
```

Azeirah, 2024-09-22:
Search for documentation feature does not work:

[image]

This is a desirable feature, so I'll have to find a workaround for it.

arulagrawal, 2025-02-12:
I also have this problem. I'm just running `nix --experimental-features 'nix-command flakes' develop github:mirrexagon/nixpkgs-esp-dev#esp32c6-idf` and then `code .` in the resulting shell to try and get everything into vscode.

johnbarnes-vi, 2025-03-12:
I also have this problem, from both the `use flake` method and the standalone shell.nix

jonsambro, 2025-07-20:
My change was merged so this issue should be fixed in ESP IDF VSCode Extension v1.11.0+.