mirrexagon/nixpkgs-esp-dev #78: How to install `esp-clang` tool?

ok-nick, 2025-03-12:
Hi, I use the [`esp-clang` tool defined here](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/tools/idf-clang-tidy.html) for `clangd` LSP support on my project. Is it possible to install it through this flake?

@mardukpill

petterhs, 2025-04-03:
I do the same and do it like this. Maybe this is useful for you.

```
{
  description = "A devshell flake for ESP-IDF";

  inputs = {
    nixpkgs.url = "github:NixOS/nixpkgs/nixos-unstable";
    nixpkgs-esp-dev.url = "github:mirrexagon/nixpkgs-esp-dev";
    flake-utils = {
      url = "github:numtide/flake-utils";
    };
  };

  outputs =
    {
      nixpkgs,
      nixpkgs-esp-dev,
      flake-utils,
      ...
    }:
    flake-utils.lib.eachDefaultSystem (
      system:
      let
        pkgs = import nixpkgs {
          inherit system;
          overlays = [ (import "${nixpkgs-esp-dev}/overlay.nix") ];
        };
        esp-clang = (
          with pkgs;
          stdenv.mkDerivation {
            pname = "esp-clang";
            version = "17.0.1";
            src = fetchzip {
              url = "https://github.com/espressif/llvm-project/releases/download/esp-17.0.1_20240419/clang-esp-17.0.1_20240419-x86_64-linux-gnu.tar.xz";
              sha256 = "DPnuoG5HuxGE8zAvesRp8qoASaa8j5mIaeF7jp8s/jY=";
            };

            nativeBuildInputs = [ autoPatchelfHook ];
            buildInputs = [
              glibc
              gcc-unwrapped
              libz
              libxml2
            ];

            installPhase = ''
              runHook preInstall
              mkdir -p $out/bin
              install -m755 bin/* $out/bin
              runHook postInstall
            '';
          }
        );
      in
      {
        devShell = pkgs.mkShell {
          buildInputs = [
            pkgs.esp-idf-esp32s3
            esp-clang
          ];
        };
      }
    );
}
```

santiagocezar, 2025-04-15:
This might be a better way to do it, by overriding the tools used by the IDF package: 

```nix
{
  inputs = {
    nixpkgs-esp-dev = {
      url = "github:mirrexagon/nixpkgs-esp-dev";
      inputs.nixpkgs.follows = "nixpkgs";
    };
  };
  outputs = { self, nixpkgs, nixpkgs-esp-dev }: 
    let 
      system = "x86_64-linux";
      pkgs = nixpkgs.legacyPackages.${system};
      # change esp-idf-esp32 to whatever is the toolchain you're using
      idf = nixpkgs-esp-dev.packages.${system}.esp-idf-esp32.override (final: {
        toolsToInclude = final.toolsToInclude ++ [
          "esp-clang"
        ];
      });
    in
      {
        devShells.${system}.default = pkgs.mkShell {
          buildInputs = with pkgs; [
            idf
          ];

          # passed as --query-driver=$CLANGD_QUERY_DRIVER to clangd
          shellHook = ''
            export CLANGD_QUERY_DRIVER=`which xtensa-esp32-elf-g++`
          '';
        };
      };
}

```