mirrexagon/nixpkgs-esp-dev #80: Overlay is currently broken on MacOS

iniw, 2025-03-27:
I hit two issues while setting up a simple hello world application:

1. `idf.py create-project` fails, which was reported in #75 
2. Espressif's compiler wrapper hits [this](https://github.com/espressif/esp-toolchain-bin-wrappers/blob/b9b42f835c44520f34f7f6523d4fb11c614f32e0/gnu-xtensa-toolchian/main.rs#L169) assert, rendering this overlay unusable since not even the simplest of programs can be compiled, causing the build to fail on the configure step.

Here's the relevant log of the second error:

```
$ idf.py build -C project/
WARNING: Python interpreter "/nix/store/la7k54xm6gjjzx5483iaki758fvby9r4-python3-3.12.9-env/bin/python3.12" used to start idf.py is not from installed venv "/nix/store/5pzkvfshy7qdqn9m9llyql3c97gvvsb9-esp-idf-v5.4/python-env"
Executing action: all (aliases: build)
Running cmake in directory /Users/sol/projects/esp-dev-test/project/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=/nix/store/la7k54xm6gjjzx5483iaki758fvby9r4-python3-3.12.9-env/bin/python3.12 -DESP_PLATFORM=1 -DCCACHE_ENABLE=0 /Users/sol/projects/esp-dev-test/project"...
-- IDF_TARGET not set, using default target: esp32
-- Found Git: /nix/store/9qjzgsf9mvdp6sfd7xyzhgrahl2qhhp6-git-2.48.1/bin/git (found version "2.48.1")
-- git rev-parse returned 'fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).'
fatal: not a git repository (or any parent up to mount point /)
Stopping at filesystem boundary (GIT_DISCOVERY_ACROSS_FILESYSTEM not set).
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
-- The ASM compiler identification is unknown
-- Found assembler: /nix/store/zy04i9551bghxn0mxnpq24n96rfqjq0d-xtensa-esp-elf-esp-idf-v5.4/bin/xtensa-esp32-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - failed
-- Check for working C compiler: /nix/store/zy04i9551bghxn0mxnpq24n96rfqjq0d-xtensa-esp-elf-esp-idf-v5.4/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /nix/store/zy04i9551bghxn0mxnpq24n96rfqjq0d-xtensa-esp-elf-esp-idf-v5.4/bin/xtensa-esp32-elf-gcc - broken
CMake Error at /nix/store/p9w37j80q80qid636hpgdwpknmrya75h-cmake-3.31.5/share/cmake-3.31/Modules/CMakeTestCCompiler.cmake:67 (message):
  The C compiler

    "/nix/store/zy04i9551bghxn0mxnpq24n96rfqjq0d-xtensa-esp-elf-esp-idf-v5.4/bin/xtensa-esp32-elf-gcc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: '/Users/sol/projects/esp-dev-test/project/build/CMakeFiles/CMakeScratch/TryCompile-yo2ba6'

    Run Build Command(s): /nix/store/cp2ql3pfclzcyzhhyj2i9rk5vnkqfwrs-ninja-1.12.1/bin/ninja -v cmTC_fbd7e
    [1/2] /nix/store/zy04i9551bghxn0mxnpq24n96rfqjq0d-xtensa-esp-elf-esp-idf-v5.4/bin/xtensa-esp32-elf-gcc   -mlongcalls -Wno-frame-address  -fno-builtin-memcpy -fno-builtin-memset -fno-builtin-bzero -fno-builtin-stpcpy -fno-builtin-strncpy -o CMakeFiles/cmTC_fbd7e.dir/testCCompiler.c.obj -c /Users/sol/projects/esp-dev-test/project/build/CMakeFiles/CMakeScratch/TryCompile-yo2ba6/testCCompiler.c
    FAILED: CMakeFiles/cmTC_fbd7e.dir/testCCompiler.c.obj
    /nix/store/zy04i9551bghxn0mxnpq24n96rfqjq0d-xtensa-esp-elf-esp-idf-v5.4/bin/xtensa-esp32-elf-gcc   -mlongcalls -Wno-frame-address  -fno-builtin-memcpy -fno-builtin-memset -fno-builtin-bzero -fno-builtin-stpcpy -fno-builtin-strncpy -o CMakeFiles/cmTC_fbd7e.dir/testCCompiler.c.obj -c /Users/sol/projects/esp-dev-test/project/build/CMakeFiles/CMakeScratch/TryCompile-yo2ba6/testCCompiler.c
    thread 'main' panicked at 'Dynconfig for target esp32 is not exist (/nix/store/zy04i9551bghxn0mxnpq24n96rfqjq0d-xtensa-esp-elf-esp-idf-v5.4/lib/xtensa_esp32.so)', main.rs:169:5
    note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
    ninja: build stopped: subcommand failed.

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  /nix/store/5pzkvfshy7qdqn9m9llyql3c97gvvsb9-esp-idf-v5.4/tools/cmake/project.cmake:564 (__project)
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!
```

I've made a repository containing the minimum amount of code for running into this (literally just the flake and an almost empty `main.c`). Just clone it, `cd` into it, enter the devshell and run `idf.py build -C project`

https://github.com/iniw/esp-dev-repro

iniw, 2025-03-27:
Reverting to the last commit before #74 fixes the issue, making me think that there's some extra workarounds needed for the 5.4 update on MacOS.

iniw, 2025-04-09:
Update on this: the PR that introduced this is actually #71 and not #74, as my previous comment states.

FredeHoey, 2025-04-09:
I'm hitting this too. @iniw Have you found a workaround?

iniw, 2025-04-09:
I just forked this repo and dropped all of the commits from that PR (and also updated to 5.4.1 while I was at it). I couldn't really figure out what the problem was and the PR didn't really add any value for my workflow, so it was easier to just remove it entirely.

iniw, 2025-04-09:
@quentinmit @'ing you since you were the author of #71. Do you have any idea what the issue could be and how to fix it?

quentinmit, 2025-04-09:
> [@quentinmit](https://github.com/quentinmit) @'ing you since you were the author of [#71](https://github.com/mirrexagon/nixpkgs-esp-dev/pull/71). Do you have any idea what the issue could be and how to fix it?

Hm. Sorry for breaking your builds. All that #71 does is to properly honor the environment variables that esp-idf requests for each tool. There's only a couple environment variables that should now be set as a result (`ESP_ROM_ELF_DIR` and `OPENOCD_SCRIPTS`), and I don't understand how either of those could have caused gcc to fail to build binaries. It also changes macOS builds to do the same thing as Linux builds (move wrapped binaries into a separate `unwrapped_bin/` directory and generate wrappers in `bin/`). I suppose that could cause problems finding files if they use a relative path.

Can you try running the compile again with `RUST_BACKTRACE=1` as suggested in the error message? It looks like the code that is complaining is written in Rust, but with just the filename `main.rs` I can't actually find the source code for it. Maybe with some function names we'll actually be able to track down the source.

I don't normally develop on macOS, but I'm giving your testcase a try and I'll update/post a new comment once I've got some more info.

quentinmit, 2025-04-09:
The backtrace I get is:

```
    Run Build Command(s): /nix/store/45wr92nx3y6r2q3l2han0x04kc34x204-ninja-1.12.1/bin/ninja -v cmTC_e1765
    [1/2] /nix/store/jjkf6wjij9j55bgiw44spx2m0rrmf2fn-xtensa-esp-elf-esp-idf-v5.4/bin/xtensa-esp32-elf-gcc   -mlongcalls -Wno-frame-address  -fno-builtin-memcpy -fno-builtin-memset -fno-builtin-bzero -fno-builtin-stpcpy -fno-builtin-strncpy -o CMakeFiles/cmTC_e1765.dir/testCCompiler.c.obj -c /Users/quentin/Documents/Personal/esp-dev-repro/project/build/CMakeFiles/CMakeScratch/TryCompile-ndn0Td/testCCompiler.c
    FAILED: CMakeFiles/cmTC_e1765.dir/testCCompiler.c.obj 
    /nix/store/jjkf6wjij9j55bgiw44spx2m0rrmf2fn-xtensa-esp-elf-esp-idf-v5.4/bin/xtensa-esp32-elf-gcc   -mlongcalls -Wno-frame-address  -fno-builtin-memcpy -fno-builtin-memset -fno-builtin-bzero -fno-builtin-stpcpy -fno-builtin-strncpy -o CMakeFiles/cmTC_e1765.dir/testCCompiler.c.obj -c /Users/quentin/Documents/Personal/esp-dev-repro/project/build/CMakeFiles/CMakeScratch/TryCompile-ndn0Td/testCCompiler.c
    thread 'main' panicked at 'Dynconfig for target esp32 is not exist (/nix/store/jjkf6wjij9j55bgiw44spx2m0rrmf2fn-xtensa-esp-elf-esp-idf-v5.4/lib/xtensa_esp32.so)', main.rs:169:5
    stack backtrace:
       0: _rust_begin_unwind
       1: core::panicking::panic_fmt
       2: xtensa_toolchian_wrapper::main
    note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.
    ninja: build stopped: subcommand failed.
```

Which led me to this source code: https://github.com/espressif/esp-toolchain-bin-wrappers/blob/master/gnu-xtensa-toolchian/main.rs

and yes, it appears to be calculating a directory relative to the "wrapper" executable. #71 causes `/nix/store/jjkf6wjij9j55bgiw44spx2m0rrmf2fn-xtensa-esp-elf-esp-idf-v5.4/bin/xtensa-esp32-elf-gcc` to now be a symlink:

```
ls -l /nix/store/jjkf6wjij9j55bgiw44spx2m0rrmf2fn-xtensa-esp-elf-esp-idf-v5.4/bin/xtensa-esp32-elf-gcc
lrwxr-xr-x 1 root wheel 111 Dec 31  1969 /nix/store/jjkf6wjij9j55bgiw44spx2m0rrmf2fn-xtensa-esp-elf-esp-idf-v5.4/bin/xtensa-esp32-elf-gcc -> /nix/store/jjkf6wjij9j55bgiw44spx2m0rrmf2fn-xtensa-esp-elf-esp-idf-v5.4/xtensa-esp-elf/bin/xtensa-esp32-elf-gcc
```
which means that the wrapper is looking (and failing) in `/nix/store/jjkf6wjij9j55bgiw44spx2m0rrmf2fn-xtensa-esp-elf-esp-idf-v5.4/` instead of `/nix/store/jjkf6wjij9j55bgiw44spx2m0rrmf2fn-xtensa-esp-elf-esp-idf-v5.4/xtensa-esp-elf/`

I think the most straightforward fix is to unconditionally generate a Nix wrapper even if there are no environment variables to set. That way the binary will be executed from the correct subdirectory. I'll see if I can turn that into a PR.

iniw, 2025-04-09:
Thanks a lot for looking into it! I actually linked the source code for the rust wrapper in the original post, but it's a bit camouflaged since it's a hyperlink 😅

quentinmit, 2025-04-09:
Oh nice. Sorry I missed that!

Your testcase passes with a PR removing 5 lines of code :) I just sent #85

iniw, 2025-04-09:
Brilliant! Works perfectly fine now.

I'm still gonna have to keep my own fork since I need a few extra python packages available in the python env (for protobuf reasons), but it's a much tidier diff now.

Thanks a lot, again.

PR #85 (2025-04-30): Fix ESP-IDF-wrapped tools on macOS
ESP-IDF has a wrapper for gcc that looks[^1] in a relative path for other files. We were previously using symlinks for binaries that don't need extra environment variables, but symlinks cause executables to be run with the wrong path in `argv[0]`. Instead, always use a Nix wrapper to ensure that executables are launched from the correct directory.

Fixes #80.

[^1] https://github.com/espressif/esp-toolchain-bin-wrappers/blob/master/gnu-xtensa-toolchian/main.rs
commit 4ac31349 (2025-04-30): Fix ESP-IDF-wrapped tools on macOS

ESP-IDF has a wrapper for gcc that looks[^1] in a relative path for
other files. We were previously using symlinks for binaries that don't
need extra environment variables, but symlinks cause executables to be
run with the wrong path in `argv[0]`. Instead, always use a Nix
wrapper to ensure that executables are launched from the correct
directory.

Fixes #80.

[^1] https://github.com/espressif/esp-toolchain-bin-wrappers/blob/master/gnu-xtensa-toolchian/main.rs
closed 2025-04-09 by iniw