mirrexagon/nixpkgs-esp-dev #90: Can't create new project, "permission denied" error

Hubro, 2025-05-05:
```
$ nix develop github:mirrexagon/nixpkgs-esp-dev#esp32c6-idf

$ idf.py create-project foo
WARNING: Python interpreter "/nix/store/25cmdl2z8mh56w0vdmm7sfsi0cb4q6s0-python3-3.12.6-env/bin/python3.12" used to start idf.py is not from insta
lled venv "/nix/store/fj8wrcavwcmv9apwsp4f9x9qfh89idf0-esp-idf-v5.4.1/python-env"
Executing action: create-project
Traceback (most recent call last):
  File "/nix/store/fj8wrcavwcmv9apwsp4f9x9qfh89idf0-esp-idf-v5.4.1/tools/idf.py", line 855, in <module>
    main()
  File "/nix/store/fj8wrcavwcmv9apwsp4f9x9qfh89idf0-esp-idf-v5.4.1/tools/idf.py", line 745, in main
    cli(argv, prog_name=PROG, complete_var=SHELL_COMPLETE_VAR)
  File "/nix/store/25cmdl2z8mh56w0vdmm7sfsi0cb4q6s0-python3-3.12.6-env/lib/python3.12/site-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/nix/store/25cmdl2z8mh56w0vdmm7sfsi0cb4q6s0-python3-3.12.6-env/lib/python3.12/site-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "/nix/store/25cmdl2z8mh56w0vdmm7sfsi0cb4q6s0-python3-3.12.6-env/lib/python3.12/site-packages/click/core.py", line 1720, in invoke
    return _process_result(rv)
           ^^^^^^^^^^^^^^^^^^^
  File "/nix/store/25cmdl2z8mh56w0vdmm7sfsi0cb4q6s0-python3-3.12.6-env/lib/python3.12/site-packages/click/core.py", line 1657, in _process_result
    value = ctx.invoke(self._result_callback, value, **ctx.params)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/nix/store/25cmdl2z8mh56w0vdmm7sfsi0cb4q6s0-python3-3.12.6-env/lib/python3.12/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/nix/store/fj8wrcavwcmv9apwsp4f9x9qfh89idf0-esp-idf-v5.4.1/tools/idf.py", line 640, in execute_tasks
    task(ctx, global_args, task.action_args)
  File "/nix/store/fj8wrcavwcmv9apwsp4f9x9qfh89idf0-esp-idf-v5.4.1/tools/idf.py", line 212, in __call__
    self.callback(self.name, context, global_args, **action_args)
  File "/nix/store/fj8wrcavwcmv9apwsp4f9x9qfh89idf0-esp-idf-v5.4.1/tools/idf_py_actions/create_ext.py", line 78, in create_new
    func_action_map[action](target_path, action_args['name'])
  File "/nix/store/fj8wrcavwcmv9apwsp4f9x9qfh89idf0-esp-idf-v5.4.1/tools/idf_py_actions/create_ext.py", line 50, in create_project
    os.rename(os.path.join(main_folder, 'main.c'), os.path.join(main_folder, '.'.join((name, 'c'))))
PermissionError: [Errno 13] Permission denied: '/home/hubro/projects/foo/main/main.c' -> '/home/hubro/projects/foo/main/foo.c'
```

```
$ eza -l --tree foo
dr-xr-xr-x   - tomas  1 jan.   1970 -N  foo
.rw-r--r-- 234 tomas  5 mai   22:41 -N ├──  CMakeLists.txt
dr-xr-xr-x   - tomas  1 jan.   1970 -N └──  main
.rw-r--r--  75 tomas  5 mai   22:41 -N     ├──  CMakeLists.txt
.rw-r--r--  45 tomas  5 mai   22:41 -N     └──  main.c
```

---

```
$ which idf.py
/nix/store/fj8wrcavwcmv9apwsp4f9x9qfh89idf0-esp-idf-v5.4.1/tools/idf.py

$ idf.py --version         
WARNING: Python interpreter "/nix/store/25cmdl2z8mh56w0vdmm7sfsi0cb4q6s0-python3-3.12.6-env/bin/python3.12" used to start idf.py is not from installed venv "/nix/store/fj8wrcavwcmv9apwsp4f9x9qfh89idf0-esp-idf-v5.4.1/python-env"
fatal: No names found, cannot describe anything.
WARNING: Git version unavailable, reading from source
ESP-IDF v5.4.1

$ readlink /nix/store/fj8wrcavwcmv9apwsp4f9x9qfh89idf0-esp-idf-v5.4.1/python-env
/nix/store/25cmdl2z8mh56w0vdmm7sfsi0cb4q6s0-python3-3.12.6-env
```

ipsavitsky, 2025-05-06:
It looks like a duplicate of https://github.com/mirrexagon/nixpkgs-esp-dev/issues/75

Hubro, 2025-05-06:
> It looks like a duplicate of [#75](https://github.com/mirrexagon/nixpkgs-esp-dev/issues/75)

Oh, I didn't see that one, it looks like the same error indeed.

I tested with a random older commit (fd227fe592f26ed4099ccd095efa66198be9cd87) and it resolved the error, so it seems like it's something new.

closed 2025-05-06 by Hubro