mirrexagon/nixpkgs-esp-dev #91: Possibly remove chip-specific shells and derivations

mirrexagon, 2025-05-08:
I don't think there is much point maintaining separate shells and ESP-IDF derivation instances for each chip, since the only difference is the set of tools. The upstream ESP-IDF install just installs all the tools for all targets. It seems more convenient to me to just use the full derivation always. 

Anyone have any opinions?

BastienGermond, 2025-08-01:
To add some context with the commit referenced above, I am proposing to add a deprecation warning until ESP-IDF 6.0 is released and then remove the chip-specif shells.

Does anyone have any reservations about this change?

mirrexagon, 2025-08-07:
I think that sounds perfect, thanks for doing that!